nightscout/AndroidAPS #1588: BUG : stuck in temp basal

robertrub, 2022-04-08:
I stopped the loop. I thought that the basal would kick in. It did not. On rhe DASH tab, I see no temp basal but on the main page, I have 0U/h (this is NOT helping my neg IOB problem either...)

I try to cancel temp basal in Actions and it doesnt work either.

[image]

[image]

[image]

[image]

robertrub, 2022-04-08:
And the logs around 13h03

[image]

robertrub, 2022-04-08:
Seems like a temp basal duration problem. It is aways 4800...

[image]

elgrupo123, 2022-04-08:
I saw the same error.

robertrub, 2022-04-09:
Seems like activation of a new DASH "solved" the problem.

robertrub, 2022-04-13:
The bug might be related to AAPS restarting. I phone battery died and when I restated it, the duration was for 4800 minutes...

[image]

Look at around 21:55

[image]

danst1peak, 2022-04-22:
I have just had the same behaviour, disabled loop as fighting a low.

Loop shows disabled but 0.0 temp basal set for 4800 mins and unable to cancel.

I didn't take a picture of the Dash tab, but that was showing no temp basal, same as Robert's picture.

[image]
[image]
[image]

danst1peak, 2022-04-22:
Logs:

[AndroidAPS.zip](https://github.com/nightscout/AndroidAPS/files/8537568/AndroidAPS.zip)

danst1peak, 2022-04-22:
I also tried changing profile to see if it reset, it didn't reset, but it did apply a different temp basal at 06:17 - even though loop was disabled.

MarkSum1, 2022-05-05:
Although changing profile doesn't always reset, if you repeat this a few times it does.

robertrub, 2022-05-07:
[image]

Things seem to work but I have this 4800 min TBR which should not be there...

avereha, 2022-06-19:
I belive this bug was fixed with https://github.com/nightscout/AndroidAPS/pull/1715
Did you still notice this issue in `dev`, after 10th of May?

robertrub, 2022-06-19:
No, I have not but I don't even remember how it had triggered. You can close the issue. If it happens again, I'll open a new one ;)

closed 2022-06-20 by MilosKozak