nightscout/AndroidAPS #1591: Incorrect SMS bolus delivered. 3.0.0.1 dev-i

elgrupo123, 2022-04-08:
Incorrect SMS bolus delivered (17.48 )
Instead of 4.0IE, 0.2IE was given.
The wrong IOB was also displayed on the NS client, it was correct on the master.

Pixel 4a/Omnipod Dash/Build 391a471b2

THX. Greetings. 

[image]
[image]
[image]
[AndroidAPS_LOG_1649448664145.log.zip](https://github.com/nightscout/AndroidAPS/files/8455194/AndroidAPS_LOG_1649448664145.log.zip)

MilosKozak, 2022-04-09:
wrong  info in NSClient can be resoled by using dev branch of NS
the wrong response was caused by clash with SMB. I'm not sure if this is resolvable

elgrupo123, 2022-04-18:
The problem hasn't happened again.
Greetings.

closed 2022-04-18 by elgrupo123