nightscout/AndroidAPS #1597: Overlapping Times in basal Tab leasing to wrong iob

Ede1975, 2022-04-10:
It seems tbrs have not been cancelled, Times are overlapping in basal tab
Possibly aaps what out of reach of dash a couple of Times during the period from 14:53 to 16:53

[image]

[image]

MilosKozak, 2022-04-13:
is it possible you changed pump meanwhile?

Ede1975, 2022-04-13:
You mean from dash to some other pump or virtual pump? No, definately not.

MilosKozak, 2022-04-14:
do you have logs prior this one?

MilosKozak, 2022-04-14:
note: where is this from
`17:02:30.229 [RxCachedThreadScheduler-53] D/DATABASE: [PumpSyncImplementation.syncTemporaryBasalWithPumpId():274]: Updated TemporaryBasal TemporaryBasal(id=28452, version=3, dateCreated=1649602950162, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=6251f4b9215b0200041e4d4c, pumpType=OMNIPOD_DASH, pumpSerial=4241, temporaryId=null, pumpId=-8976659072097442112, startId=null, endId=-7142919318749436032), timestamp=1649602949159, utcOffset=7200000, type=NORMAL, isAbsolute=true, rate=1.5, duration=1800000)
`

Ede1975, 2022-04-14:
> do you have logs prior this one?

No, unfortunately not. I am still on the same Version, did not happen again. In case it occurs again i will add new logs.

emmatovar27, 2022-04-20:
Hi There! I have the same issue. This happens every time the TBR is set for 30 min. The rest of TBR looks to work fine.
[image]
[AndroidAPS_LOG_1650416536289log.zip](https://github.com/nightscout/AndroidAPS/files/8517378/AndroidAPS_LOG_1650416536289.log.zip)

 Let me know if there is anything additional information is need. 

I'm using the dev branch from Philoul to test Autotune.

vanelsberg, 2022-04-25:
Could not confirm but now seeing similar, quite a lot of this overlapping TBR's at certain times.
Am running DASH while building from Philoul's Autotune PR (merged with latest dev).

Will switch back building form current dev to see if it reproduces with some helpfull logging
@MilosKozak Any suggestions on how to "debug" this. Assuming at least logfiles - anything else to watch for?

[image]

nicoleloop, 2022-04-25:
I am experiencing the same issues with my Dash Pods, as well as terrible bluetooth connections with the last 4 pods. In the range of the lowest, 51% to the highest being 63% connection quality.

MilosKozak, 2022-04-26:
@vanelsberg do you have logs from this time?

robertrub, 2022-04-26:
I started looking and found that I have the same bug

[image]

vanelsberg, 2022-04-26:
> ... try dev and provide logs
Busy trying to repro and have logging/screenprints. Expect real soon now...

vanelsberg, 2022-04-26:
@MilosKozak Omipod DASH, now running latest dev builds from yesterday/today. Also seeing overlapping TBR.
(Note that I am running an outdated version of NS - do not think this is relevant?)

Have 2 sets of logging with additional screenshots
Also adding .zip if the complete AAPS logging for today:

[1.zip](https://github.com/nightscout/AndroidAPS/files/8566414/1.zip)
[2.zip](https://github.com/nightscout/AndroidAPS/files/8566415/2.zip)
[morelog.zip](https://github.com/nightscout/AndroidAPS/files/8566416/morelog.zip)

--------------------------------------------------
[image]
[image]

vanelsberg, 2022-04-26:
Did not come to checking logs myself, maybe tomorrow...
If you need additional info of specific testing, no problem, let me know...

robertrub, 2022-04-26:
It is a killer... I'm not sure going on using _dev_

[image]

[image]

[image]

[image]

robertrub, 2022-04-27:
I went back to _Master_ 3.0.0.1 and overlapping TBRs stopped.

vanelsberg, 2022-04-27:
Screenprint showing overlap + logging (lots - had logging of events enabled - switched this off now)
[20220427.zip](https://github.com/nightscout/AndroidAPS/files/8570101/20220427.zip)

robertrub, 2022-04-27:
After 12h of use, I confirm that the TBR overlapping issue does not exist in _master v3.0.0.1_.

mick711, 2022-04-27:
[image]
[image]
Last dev i 
[Uploading AndroidAPS.log…]()

nicoleloop, 2022-04-27:
> [Screenshot_20220427-200812044]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F441734%2F165591901-2e3f638e-c5e5-4dfd-b8b0-5fb0d90467c2.jpg&data=05%7C01%7C%7C5024ee6af1194ee636a308da2879867b%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637866799664227550%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=RhK6M4cbf94%2BAg6vDRRDv1u%2BLvGmvHI9r1%2FXnv4QbsI%3D&reserved=0>
> [Screenshot_20220427-200823761]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F441734%2F165591913-1e9528f2-e7a2-4e71-b9be-9ac884803096.jpg&data=05%7C01%7C%7C5024ee6af1194ee636a308da2879867b%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637866799664227550%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=W65pNoxF%2F1b1w2hOgZR3elvP4uXFzwyLFBgsjHkZ4vk%3D&reserved=0>

I was woken up 3 times last night by low bg events alarms. Does anyone have any idea what's happening? Can't we just revert the last month of dev updates and start over? I am worried that this is going to cause something worse than just having alarms go off, like not being any to wake up because AAPS hadn't recognized or recorded 118 +111 minutes of basal! My low event last night was so pronounced that I walked into the wall of my own bedroom and then crumpled on the floor. My bg registered 26mg/dL when I checked it with my meter. At this point I would support reverting all of the last month's worth of commits on the Dev branch, as this is a very dangerous bug!

jgslade, 2022-04-28:
I got an overlapping basal today. Looking into what was going on at the time,  it looks like a tbr was set,  an smb was delivered,  an automation was triggered setting a temp target, then another tbr was set. 

[image]

[AndroidAPS._2022-04-27_00-00-09_.4.zip](https://github.com/nightscout/AndroidAPS/files/8579512/AndroidAPS._2022-04-27_00-00-09_.4.zip)

nicoleloop, 2022-04-28:
I have yet to look in the log files to verify this info, but right off the top of my head- temp basals are affected in this way, but not SMBs.

[Screenshot_20220427-215332_AndroidAPS]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F29897815%2F165676730-be897c6f-7096-4623-b494-c727b1ff8a93.jpg&data=05%7C01%7C%7Ccb0a6113f4fd4702c49a08da28d03de8%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637867172105993931%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=QS4fY%2BOufpBCTIIuba1Y9cKJ21Cmbm2Ya7sTwYUNEmQ%3D&reserved=0>

tswire, 2022-04-28:
Has anyone confirmed if this truly results in incorrect IOB?  More importantly, if that IOB is used in the predictions and AAPS dosing decisions?  I think it's very obvious that something is "wrong" based on the treatments tab for TB.  However, I think we need to know immediately and urgently if this could be dangerous, as seen by a few examples of people going low.

vanelsberg, 2022-04-28:
> I was woken up 3 times last night by low bg events alarms.
Same here a number of times.
As far as I could tell the much-argued negative IOB problem: too much insulin causing high negative insulin buildup that AAPS will try to fill in as soon as BGs are rising - followed by another low as logic dictates. This could be a side-effect?

Not sure, but I do think we need confirmation that the TBR problems we see are not affecting IOB calculations. I could switch back to release version, but then I would not be able to help document/analyse/solve this problem. So will stay on dev as long as I think is safe enough.

Ede1975, 2022-04-29:
> Has anyone confirmed if this truly results in incorrect IOB?  More importantly, if that IOB is used in the predictions and AAPS dosing decisions?  I think it's very obvious that something is "wrong" [...]

From my point of view its obvious that it affects iob calculations. It counts (negative) insulin twice when Times overlap. If you remove an overlapping treatment (which is Not 0 yet) , the calculated iob changes instantly. Same for positive overlaps of course.

vanelsberg, 2022-04-29:
This issue is labeled "DASH".
Is it only for Omnipod DASH pump the overlapping TBR is occurring?

MilosKozak, 2022-04-29:
i improved logging https://github.com/nightscout/AndroidAPS/commit/2d653c67ef9c32201fe0e28897af4904f9c423cb
i'd need someone to reset AAPS DB (with zero IOB COB to be safe) and try latest dev
Provide logs if you see this again

vanelsberg, 2022-04-29:
> i improved logging [2d653c6](https://github.com/nightscout/AndroidAPS/commit/2d653c67ef9c32201fe0e28897af4904f9c423cb) i'd need someone to reset AAPS DB (with zero IOB COB to be safe) and try latest dev Provide logs if you see this again

@MilosKozak About to build, reset database and then watch for repro.
Expect logging as soon as I have some.

vanelsberg, 2022-04-29:
> i improved logging [2d653c6]...

Provide logs if you see this again
Any special log settings to eneable?

MilosKozak, 2022-04-29:
PUMP*

mick711, 2022-04-30:
[image]
[image]
I try to find différence Master and Dev

vanelsberg, 2022-04-30:
> PUMP*
@MilosKozak 
Yesterday evening install latest dev, did database reset: until now (24hrs) no overlap.
(detail: pump connection is 100%, so no fail/retries on commands - will test later if any relation with failures?)

(Expect logging as soon as I see overlapping occur)

emmatovar27, 2022-04-30:
> 

Same Here! :)

TBR is not overlapping anymore! 

Thanks @MilosKozak 

[image]

nicoleloop, 2022-04-30:
Wonderful! Thanks Milos! 😁

[TempBasalRates]<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F39507448%2F166122364-6af6dc19-b192-437b-a854-db1e265a085e.png&data=05%7C01%7C%7Cc0ccc8eeb36340b31c7108da2aebdbf3%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637869489736324622%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=CQ%2F0wvcI6JSLnEp2WWSN17rr6Wk2YthdmmIs40a3YVI%3D&reserved=0>

vanelsberg, 2022-05-01:
After 2 days, with new build and database reset, still not seeing anything on TBR overlapping.
Wondering if database could be a factor in causing this?

ArthurusDent, 2022-05-01:
@vanelsberg If this is the same bug I saw, it might only happen when disabling the pump für 2 h. Didn't notice it with shorter periods but I might just not have noticed. Could you maybe test 2 h, too?: Disabling the pump for 2 h, waiting for the algorithm to run once and then re-enabling the pump again?

MilosKozak, 2022-05-02:
i believe it was corrupted data

vanelsberg, 2022-05-03:
> .. i believe it was corrupted data

@MilosKozak Was thinking the same. Had been testing autotune on separate branch. Now on dev latest dev branch with databases reset, but still:
Overlap again past night (see 20220503, 02:59h, 04:59h,)

Coincided with two G6 abnormalities where it started alerting low (while initially on target) needing carbs (partly unregistered). 
Not sure if this could be related. See screenprints G6 and FSL I have additionally active for testing/comparance.

**All logging/info:**
[20220503.zip](https://github.com/nightscout/AndroidAPS/files/8608440/20220503.zip)

**Screenshots;**
[image]
[image]

MilosKozak, 2022-05-03:
if I see correctly the fixing commit is not included in autotune branch
https://github.com/nightscout/AndroidAPS/commit/1684b88270b34d0205f3e10ad3109e2319802532

tswire, 2022-05-03:
> if I see correctly the fixing commit is not included in autotune branch [1684b88](https://github.com/nightscout/AndroidAPS/commit/1684b88270b34d0205f3e10ad3109e2319802532)

But he said this overrlapping was from updated Dev (as of yesterday) and after resetting DB?  So that commit was 20 days ago and not sure how that fits in to this?

jgslade, 2022-05-03:
I got some overlapping tbrs last night.  Using the issue with improved logging and reset the db after updating.

The 10:34-11:04 and 10:59-11:09 one is interesting because it's the same tbr value.  

[image]

[AndroidAPS._2022-05-02_21-47-20_.1.zip](https://github.com/nightscout/AndroidAPS/files/8610629/AndroidAPS._2022-05-02_21-47-20_.1.zip)

vanelsberg, 2022-05-03:
> if I see correctly the fixing commit is not included in autotune branch

FYI: Not using autotune branch but latest "dev" #7d93a2ceef

kristinsmotz, 2022-05-03:
yesterday found the first overlapping TBR's I have noticed. We're on Dash, and testing Dynamic ISF on the second latest version (Dynamic ISF beta 1.6.1). What was interesting to me was that it occurred in the same timeframe where I encountered the first re-occurance of my other issue (https://github.com/nightscout/AndroidAPS/issues/1667 where bolus doses (and SMB's) occasionally don't show up in the NSClient phone, only in AAPS phone and on our nightscout site), since I cleared out memory and cache related to NSClient on my phone a few days ago. Could these issues be related in any way? Since clearing database (or in my case clearing phone memory) appears to help for a period of time, could it have anything to database structure/save data file size?

avereha, 2022-05-04:
Can you please to disable Nightscout sync and check if the issue is still there?

vanelsberg, 2022-05-04:
> Can you please to disable Nightscout sync and check if the issue is still there?

Would setting NSClient to "Paused" (checkbox) do?

avereha, 2022-05-04:
yes

vanelsberg, 2022-05-04:
First results. Will run this for some time trying to "leave it alone" and post new logs here once overlapping happens.

Did update to new build for #1691 around 21:45h (?).
Now seeing this. Not sure: why are these most recent TBRs suddenly with only a starting time?

See "22:11-22:41 1.75U/h 30 mins":
It was canceled (also on DASH history) but somehow this was not registered to AAPS?

[image]

** Log files + screenshot **
[20220504.zip](https://github.com/nightscout/AndroidAPS/files/8627412/20220504.zip)

jgslade, 2022-05-04:
I turned off nightscout syncing this morning,  checked this afternoon and I've still got overlapping basals.
[image]
[AndroidAPS._2022-05-04_00-00-00_.2.zip](https://github.com/nightscout/AndroidAPS/files/8627781/AndroidAPS._2022-05-04_00-00-00_.2.zip)
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8627782/AndroidAPS.log)

vanelsberg, 2022-05-05:
Lots of TBR overlap (see 00:00 ... 03:00h)?

[image]

** Logfiles & screenprint **
[20220505-1.zip](https://github.com/nightscout/AndroidAPS/files/8629169/20220505-1.zip)

MilosKozak, 2022-05-05:
looks like AAPS is updating old records.
i added more logging to dev https://github.com/nightscout/AndroidAPS/commit/d131ceb4bd52e85d92650f6dff26692f73d822de
please again update and reset db + pause NSClient

MilosKozak, 2022-05-05:
maybe remove old logs too and once this happen upload all available logs (if there are not too many)

vanelsberg, 2022-05-06:
@MilosKozak @avereha 
Have been running commit #3d9b08e5c2 with NS paused for 24 hrs: no overlap.
With it, BG's are steady as rock again. Maybe to early to conclude NS is the problem. Wil wait for another 12 hrs. Then "unpause" NS to see what happens. Expect logging as soon as overlap happens again.

vanelsberg, 2022-05-08:
> Have been running commit #3d9b08e5c2 with NS paused for 24 hrs: no overlap.

Still not seeing any overlap happening. Tried syncing NS by temporarily unpausing NS until synced (then pause again):  no overlap.

Switched to new phone yesterday evening (read started with new database).
Now running for about 24 hours. Will enable NS syncing to see if this results in overlap occurring.

vanelsberg, 2022-05-09:
@avereha @MilosKozak More logs.

Was unable to reproduce TBR overlapping with NSClient paused. Temporary unpausing NSClient to catch up syncing (and then pausing again when don) did not give problems.

Switched to a new phone yesterday, so empty database, clean OS, clean install.
Today, unpaused NSClient and TBR overlapping started occurring again. Looks like NS syncing introduces/relates to this problem?

Screenshots and logging attached below. Please let me know when I can assist any further in finding the problem!

[20220509.zip](https://github.com/nightscout/AndroidAPS/files/8654463/20220509.zip)

MilosKozak, 2022-05-10:
hopefully fixed by #1715 
open new issue if the problem persists

closed 2022-05-10 by MilosKozak

ArthurusDent, 2022-05-10:
It looks like #1715 only fixes Dash. I'm on regular 3.0.0.1 and Medtronic and I'm seeing this bug. Could somebody please check the Medtronic plugin? Thanks!

MilosKozak, 2022-05-11:
@ArthurusDent open new MDT specific issue with logs and info