nightscout/AndroidAPS #1598: Buttons disappear and sometimes don't reappear - dev-i

JohnKitching, 2022-04-10:
Carbs, calculator, insulin, treatments buttons. 
I'm guessing they've been set to hide at times to maximise screen space. When connecting to pump perhaps?
I had an issue today that pump battery died so my buttons never reappeared (I'm guessing it was waiting for a pump disconnect event?). 
This was quite a problem as I could have really used the calculator to tell me what I needed from my backup pen.

Andries-Smit, 2022-04-10:
By design, the buttons are only available when you can use them.

JohnKitching, 2022-04-10:
Don't think I've ever needed them more than I did today to calculate a manual dose.

Andries-Smit, 2022-04-10:
Carbs and CGM buttons are available  when pump is offline.
Theoretical the insuline button can be used without the pump, as it is has a "do not bolus, record only" checkbox.
By removing the buttons it makes it realy clear to the users that there the regular therapy is not available.

JohnKitching, 2022-04-10:
All the buttons were hidden for me today. 

If you don't want to show the buttons how about providing another way to access the functionality?

Andries-Smit, 2022-04-11:
To understand why the buttons are hidden, we can look in the code https://github.com/nightscout/AndroidAPS/blob/dev/app/src/main/java/info/nightscout/androidaps/plugins/general/overview/OverviewFragment.kt#L533
- Profile should be selected
- If the pump stores carbs info, you can not enter carbs when it is not connected; some Dana pumps do support it

Making the calculator available when the pump is not available is a niche function, and have to discuss effort / vs value with the team.

JohnKitching, 2022-04-11:
It's not a biggie to me. I rarely use the buttons. I was just reporting a problem. I thought that was the point of testing dev. Sorry.

Andries-Smit, 2022-04-11:
@JohnKitching, no need for apologies; I am happy to help

closed 2022-04-11 by MilosKozak