nightscout/AndroidAPS #1602: 4 Pod faults in 24 hours - Has EROS communication changed?

Technixx, 2022-04-10:
Hi, 

My son is using AAPS v3
Unihertz Jelly2, Android 10 
I use Orangelink Pro and Original with firmware 3.2. - fresh batteries

I have now had 2 screamers in 24 hours. 
I have problems with activation and pod unreachable 2 pods failing multiple times during activation with fault: 049 | 0x31 | Incorrect pod state for command

I am now at this time 10 April at 23:17 - on the 5th pod - still getting errors but managed to activate this one.

It seems like it can fail multiple times, then get the correct "GET STATUS" for a few minutes then back to pod fault.

It has started with the 3.0.0.1 and the new Lot version for Eros. I can't tell if it's either one of them.

ninelore, 2022-04-11:
As mentioned on Discord: Verify the batch with the original PDM to rule out a bad batch.

Technixx, 2022-04-20:
I have tried to use 3 more pods now, without these problems. I have scrolled through multiple logs from, and not found anything unusual. I am still unsure about why this incidents occurred and why they happened in a row. But it might actually have been just bad pods. I know thy try to mix these pods to minimize this risk, but it seems like it might have been a box of multiple bad pods coincidentally. I will reopen the issue if this happens again.

closed 2022-04-20 by Technixx