nightscout/AndroidAPS #1604: MINOR / WIDGET / Color difference for mid-high BGs

robertrub, 2022-04-11:
AAPS doesn't use the same color on the app and widget for mid-high BGs

App

[image]

Widget

[image]

osodebailar, 2022-04-11:
The color from the widget is taken from the default color.xml. 
This are the colors for the light mode in aaps.
In the light mode yellow is not possible , because yellow on white background is hard to see. 
Aaps itself is in darkmode and has another color.xml table.
The widget itself has no possibilities for dark or light mode. It's not implemented by google until now.

MilosKozak, 2022-04-11:
could we create special color names for widget which would match dark mode?

osodebailar, 2022-04-11:
@MilosKozak : yes, I will have a look. I will prepare a PR this evening

osodebailar, 2022-04-11:
PR is ready: https://github.com/nightscout/AndroidAPS/pull/1605

closed 2022-04-11 by MilosKozak