nightscout/AndroidAPS #1609: Absolute Insuline negative

kamlox, 2022-04-11:
Hello,

I encountered a bug where my absolute insule was negative. What should not be possible.
IOB was -2.5 it seems.
Profile has 0.9u/h

Context:
This happened the night/morning after a day of drinking, and drinking a little too much.
First hypo in the night that I didn't feel, AAPS managed then hypo when I woke up and I had never seen an IOB that low.

kamlox, 2022-04-12:
Same problem today, but no alcohol.
There is no way my BIO is that low.
I've had this behavior for a week since I upgrade in the last dev.

[image]

[image]
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8471303/AndroidAPS.log)

kamlox, 2022-04-12:
> do yopu have screenshot of treatment - bolus & basal tab from the same time? some time before and after

Do you need more infos?

[image]
[image]

MilosKozak, 2022-04-12:
note. this seems to be the problem
[image]

mick711, 2022-04-12:
If i remember #robertrub have open an issue with the 4800mins

kamlox, 2022-04-12:
> Its not stopping Zero temp and continues counting negative Insulin, same as in #1597. Even without the 4800 minutes it would go wrong

and like you I use the dash

MilosKozak, 2022-04-13:
fixed by https://github.com/nightscout/AndroidAPS/commit/b756a870887015f8af609e5915f2c03f47ee8518

closed 2022-04-13 by MilosKozak

robertrub, 2022-05-07:
@MilosKozak Sorry but it is NOT solved. It still happens in dev-j. See issue #1706 

AndroidAPS 3.0.0.2-dev-j 
Build: 16d39421b7-2022.05.07-14:30