nightscout/AndroidAPS #1610: SMS Communicator Not Sending Response in Dev i but still works in Master (tried on multiple phones)

cygrahampearson, 2022-04-12:
SMS communicator is not sending a response (but will still implement commands). 

This problem first began when I switched from @tim2000s main dynamic ISF branch to his dev. I briefly switched back to master and it worked fine. I then switched to NS dev G, H and then I and its not working. I've tried on multiple phones (S9, Jelly 2, Atom and Not 10) and its not working on any of them. Master still works fine. 
--------------
- Time: 8:40pm I sent five different SMS commands
-AndroidAPS 3.0.0.1-dev-i
-BUILD: 671127673-2022.04.05-12:55
-logs and screenshots attached
[AndroidAPS_LOG_1649724181836.log.zip](https://github.com/nightscout/AndroidAPS/files/8468679/AndroidAPS_LOG_1649724181836.log.zip)

[image]
[image]
[image]
[image]

Please let me know if there is anything I can do to help get you some useful data. 

Thank you, 

Coleby

MilosKozak, 2022-04-12:
problem is on master. AAPS cannot send SMS even we see it's trying it

MilosKozak, 2022-04-13:
what do you see in AAPS permissions in android setting?

cygrahampearson, 2022-04-14:
@MilosKozak  Would it be helpful to send a video showing SMS commands sending texts in Master and then switching to dev ? I can show the settings and permissions and switch back to master to show. 

I've attached pictures of permissions, please let me know if you see anythin
[image]
[image]
[image]
[image]
[image]
[image]
g

MilosKozak, 2022-04-14:
there is no change in SMS code between master and dev-i except some text formating

MilosKozak, 2022-04-14:
try it on another phone

jumalakarhu, 2022-04-14:
I can confirm the same with Sony xperia xz1 compact.  sms command in dev is not functional. Sms is read in aaps and respons issued  (seen in sms log in aaps), but sms is not actually sent from phone. Works fine when going back to master. Could it be something about permissions not properly inherited/handled (?) in dev from master? Would be worth testing with a fresh install of dev, but I do not have a spare android 9+ at the moment.

cygrahampearson, 2022-04-14:
@MilosKozak I've tried on Samsung note 10, Jelly 2, Atom and S9 all with no luck. They all work fine with Master. 

I'll take a video performing SMS commands and then switching from dev to master to new dev to show how its working/ not working and will pull some more logs. 

Is it helpful for me to adjust the log settings to add or remove certain categories?

It looks like @jumalakarhu can confirm what I am seeing as well.

cygrahampearson, 2022-04-15:
@MilosKozak Here's a video of me attempting and failing to respond to SMS commands in dev-i and then installing master to show them working and then to dev h. 

https://drive.google.com/file/d/121tXlCh4dX9zuVP6RbcVSVzWT621wZWz/view?usp=sharing

Im updating now to latest dev to see if there are any changes

rosseland, 2022-04-16:
Same issue here, dev-i, Huawei p20pro, android10

shafiro, 2022-04-21:
Had the same problem with Galaxy S21+. After upgrading to Android 12, SMS functionality is back to normal.

MilosKozak, 2022-05-19:
resolved by https://github.com/nightscout/AndroidAPS/commit/6f358bc0c80b902b0abbf6b0dbd77a88b035ee24

closed 2022-05-19 by MilosKozak