nightscout/AndroidAPS #1615: Insight Pump prime not reset canular time/date

chrisbev, 2022-04-13:
When an Infusion site change was made at the pump then the canular date/time was not made/updated in AAPS.
AAPS 3.0.0.1 dev-i.  Insight F/W V3.00.17. Around 20:57hrs.  Log and screenshots attached.  Treatment tab showing pump disconnect and connect with no prime in-between.

[AndroidAPS._2022-04-13_00-00-01_.7.zip](https://github.com/nightscout/AndroidAPS/files/8484867/AndroidAPS._2022-04-13_00-00-01_.7.zip)
[image]
[image]

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

chrisbev, 2022-04-14:
Cartridge and Tube changes are actioned in AAPS OK.

chrisbev, 2022-04-14:
NB Cartridge change does update.

Philoul, 2022-04-14:
How many U of insulin is injected when you prime Canula from your Pump?
I'll try to prepare a version of AAPS with more log to analyse what could be wrong in your situation... (I made exactly same test on my side with my pump, same firmware, and it works)
[image]

chrisbev, 2022-04-14:
Because I was going to control pump from pump buttons I didn't want AAPS to interfere or get confused so thought it would be best to disconnect (can't remember if there was basal running at the time?). Canular prime 0.7U.  I will try without disconnection.

Philoul, 2022-04-14:
> Canular prime 0.7U

Ok, so it should work.
Let me make a dedicated version with more log  in insight driver, to be able to trace and understand what could be wrong in your case (I'll try this evening).

chrisbev, 2022-04-14:
Hey, don't worry if only me hardly a big deal, it doesn’t coinside very often and easy workaround I mainly cannula change through AAPS you are busy with other things, it only started as a question on TDD which is now confirmed as OK.

Philoul, 2022-04-14:
@chrisbev It's probably a stupid question, but have you checked if in Insight preferences, "Log site changes" setting is enabled?
[image]

chrisbev, 2022-04-14:
@Philoul you are kind never a silly question you suspected right it was set off, don't know how/when.  Thanks for investigating can only say sorry to have troubled you.

closed 2022-04-14 by chrisbev

Philoul, 2022-04-14:
Cool.
You probably set this preference to off with aaps v2.8 (or earlier).
A site change was always recorded with a reservoir change (this bug was only for Insight pump with firmware v3).
I fixed this issue in Insight driver of aaps v3...