nightscout/AndroidAPS #1619: [DEV] Combo: Cancelled TBR alarms

Der-Schubi, 2022-04-14:
I'm on dev branch and as I already mentioned in Discord, I'm getting frequent cancelled TBR alarms on my Combo.
AAPS 3.0 doesn't have this problem.
These alarms are occurring in the middle of pump communication and I have to confirm them manually on the pump.
There are sometimes two alarms a day, sometimes five in only one hour, but I get them constantly.

I tracked the problem down to commit [Combo: cleanup code](https://github.com/nightscout/AndroidAPS/commit/ad8c2297a01a9165f1e9babc29f63bb9bf46e8f4), with this commit reverted, I'm running without any cancelled TBR alarms for 10 days now.
As this commit doesn't change much but introducing the logging.AAPSLogger, I'm unable to find a reason for this.
The logs seem to point to timeouts reported by ruffyScripter: [AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8489246/AndroidAPS.log)
Any Idea what could cause this?

cominixo, 2022-04-21:
Can confirm this, reverting the commit also works.
Was also experiencing more frequent disconnects from the pump (probably due to the cancelled TBR alarms)

Der-Schubi, 2022-05-02:
The reason for the alarms might be a timing issue within RuffyScripter. AAPSLogger seems to be slower than the old slf4j logger.
Removing all aapsLogger.debug lines ("Ruffy says ...") in RuffyScripter.java solves the problem.
What could be a possible solution? Ignoring? Disabling debug logging in RuffyScripter??
Or is there a simple way to disable debug logging for dev builds?

MilosKozak, 2022-05-03:
@jotomo could be Ruffy so sensitive?

jotomo, 2022-05-03:
@Der-Schubi Nice job bisecting the issue, much appreciated! :+1: 
@cominixo Thanks for confirming!

The change causing the issue is that previously the "ruffy says" logging was using the _trace_ level, which was not active by default. Since AAPSLogger has no _trace_ level, the refactoring changed it to _debug_, thereby enabling rather verbose logging. The amount of added logging then apparently causes the issue. I've opened PR #1688, which simply removes the those logging calls, which aren't useful to be enabled all the time. This then resolves the issue as demonstrated by @Der-Schubi

Der-Schubi, 2022-05-03:
Thanks a lot @jotomo 
Didn't notice the change of log level.
Closed :)

commit 00645f45 (2022-05-03): Merge pull request #1688 from jotomo/combo-ruffy-chattyness

Fix #1619 by removing unnecessarily chatty ruffy logging.
commit 8c23f4f6 (2022-05-03): Fix #1619 by removing unnecessarily chatty ruffy logging.
closed 2022-05-03 by Der-Schubi