nightscout/AndroidAPS #1621: OLD BUG : Activate/Deactivate pod buttons are lit

robertrub, 2022-04-14:
This bug was solved but is back once more. **Both** pod _activate_ and _deactivate_ buttons are turned ON (lit, not grayed) all the time (when pod is activated and when no pod is activated).

[image]

[image]

robertrub, 2022-04-14:
Seems similar to #1614 but mine was with DASH.

Andries-Smit, 2022-04-14:
@robertrub I quickly checked. Previously it was caused by the styling changes. The last build seems fine on my test phone but does not have any active pods.

robertrub, 2022-04-14:
I have both lit with active and no active pods.

CaptainBalou, 2022-04-14:
Here I've got the buttons correct. During the pod was on and after it was deactivated. No trouble.

I don't know how to add a screenshot on my phone but I would have one I could add later if needed.

CaptainBalou, 2022-04-14:
I'm also on dev-i at the moment but built 7 days earlier.

robertrub, 2022-04-15:
A week ago, it was working ;)

Thomas-Wolff, 2022-04-15:
I can also confirm this, I have already opened a new issue #1614 two days ago.
Greetings Thomas

elgrupo123, 2022-04-15:
I can also confirm the error in the last build (1ba67d9f7).
In build before (391a471b2) everything is OK.

CaptainBalou, 2022-04-19:
@robertrub 
> A week ago, it was working ;)

After I fixed a build issue in Android Studio I was able to build the app again. After using this build 4398a433c-2022.04.18-23:21 I can confirm that the buttons are both active here as well.

osodebailar, 2022-04-19:
> @robertrub
> 
> > A week ago, it was working ;)
> 
> After I fixed a build issue in Android Studio I was able to build the app again. After using this build [4398a43](https://github.com/nightscout/AndroidAPS/commit/4398a433c3fbf257597abe2a52c734088def94ed)-2022.04.18-23:21 I can confirm that the buttons are both active here as well.

Please try it again with new version of dev. In simulation it seems to work

elgrupo123, 2022-04-19:
@osodebailar 
The latest (49af54) works again. Unfortunately, the graphic below is much too high and the font is too small
[image]
.

robertrub, 2022-04-19:
Sorry, on vacation till next tuesday with no PC. Will test in a week.

CaptainBalou, 2022-04-19:
@osodebailar 
> Please try it again with new version of dev. In simulation it seems to work

For me all looks good again. My pod is online and button to activate is off, button do deactivate is on. Perfect.

robertrub, 2022-04-26:
Solved :)

closed 2022-04-26 by robertrub