nightscout/AndroidAPS #1627: "Active pod" after EROS pod error prevents starting new pod

jalbers01, 2022-04-17:
With AAPS version 3.0.0.1-stable / Omnipod EROS, I regularly observed these errors (several times from 04/04/2022 until now):

1) Pod can be deactivated and removed from system with AAPS menu tools, but afterwards I get information like "pod-activation in progress" (in Omnipod-menu), and at the same time a pop-up-message, no pod active. 
It's not possible to discard this (not existing) pod-activation. No new pod can be acitvated, the systems assumes there is a pod connected which is in the initalization process. Trying to activate a pod aborts latest before setting of the cannula as a screamer-pod. 
2) This makes the AAPS app impossible to use. The only workaround seems to be to load a new setup file, which replaces the existing setup. 
That was only possible with an old setup file from AAPS 2.8.2 version. One pod can be started without problems, but after removing this pod the problem starts again with 1) . 
3) All trials to create a new setup file with AAPS 3.0.0.1 failed, leads after loading of this file to 1) without possibility to start any pod. With these files not possible to "kill" the virtial incatiove pod from the system.

samspycher, 2022-05-21:
@jalbers01 could you send a log file for the time range of such a pod deactivation?

jalbers01, 2022-05-22:
Unfortunately, I have no log files anymore. I replaced 3.0.0.1 with 2.8.2.1 which runs w/o any issues. But I didn't save any logfiles before downgrading.

samspycher, 2022-05-23:
@MilosKozak since we have no reproduction for this with logs, it may make sense to close it? I cannot reproduce this on my side.

closed 2022-05-23 by MilosKozak

majatocaha, 2022-05-28:
Just encountered the same situation and stumbled upon this github issue. Sent you guys the log files and a brief description of what happened.

MilosKozak, 2022-05-28:
@majatocaha email logs are not working properly. create an issue