nightscout/AndroidAPS #1633: Automation preconditions not working

angielizzy, 2022-04-18:
Using AAPS version 3.0.1 when running this automation it is triggering every time blood glucose is about 14.0mmol despite already having a different profile switch set. This didn't happen in previous versions of AAPS
Expected behaviour - if currently running a profile switch above 100 then the automation won't trigger due to the precondition
Current behaviour - automation triggers no matter what profile switch is running so the profile will keep getting changed to 115% with no way to stop it without disabling the automation.
[image]

--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-04-26:
do you have a screenshot to prove that? (bottom of automation tab)

angielizzy, 2022-04-28:
Unfortunately I don't but it seems have done same thing today with an 80% profile switch from an automation when it's already running an 80% profile switch so I will have a look at my daughter phone when she gets home from school and take the relevant screen shots.

closed 2022-05-10 by MilosKozak