nightscout/AndroidAPS #1648: The last layout changes are inconvenient for small devices

swissalpine, 2022-04-19:
Yes, I use a smartphone with a small screen, but I think I'm not the only one. From there, you should be careful with layout changes that take up space and don't add functionality. That's what the changes Material Card View and Material Button do.
Maybe it is possible to change the margin-top for small devices to 0 dp in the Material Card View (I changed it from 4 dp to 1 dp so I can see the graph again), how to adjust the material buttons is beyond me. Please revert this changes!
New Matieralbuttons:
[image]
[image]

----

Old buttons:
[image]
[image]

Please undo at least the material button commit!
And have an eye on the impact on small Screens as well for future layout changes (btw I liked the old layout!). Thanks!

osodebailar, 2022-04-19:
@swissalpine  We can change the style to smaller text , so it fits again .

Andries-Smit, 2022-04-19:
There are 2 types of users, users with a personal phone running AAPS, do have, in general, a large screen.
And users with a designated device, which is quite often as small as possible.

To cater for both, we can easily distinguish using qualified resources. and then we can tweak for smaller screens, for example, setting some padding/margins/corner radius to 0.

https://developer.android.com/guide/topics/large-screens/support-different-screen-sizes

Philoul, 2022-04-20:
Even if it's not a small device, I think marging in material button should be reduced (close than half of the width is lost...)
[image]

osodebailar, 2022-04-20:
> Even if it's not a small device, I think marging in material button should be reduced (close than half of the width is lost in overview) [image]

Changed the former workaround margin:
https://github.com/nightscout/AndroidAPS/pull/1652

swissalpine, 2022-04-21:
Thanks for the effort and finding solutions!

closed 2022-04-21 by swissalpine