nightscout/AndroidAPS #1659: Hello everyone, the latest DEV support GlucoRx Aidex as blood sugar use, my country has this blood glucose meter, please how do I start setting up to use

qq254344379, 2022-04-21:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

mick711, 2022-04-21:
Use xdrip with this cgm. Post on Facebook or Discord for help 😉

qq254344379, 2022-04-21:
> 将此 cgm 与 xdrip 一起使用。在 Facebook 或 Discord 上发帖寻求帮助😉

Discord, I can't sign in. I'll try Facebook

MilosKozak, 2022-04-21:
@andyrozman

qq254344379, 2022-04-21:
I'm willing to test it for that

andyrozman, 2022-04-21:
This is not yet available. GlucoRx app that will support this was not yet released.

@mick711 Aidex support for Xdrip is also in the works, but was not yet merged. But until GlucoRx app supports it you won't be able to test it.

mick711, 2022-04-21:
Thanks for this information ;) , i use dexcom g6

qq254344379, 2022-04-22:
> This is not yet available. GlucoRx app that will support this was not yet released.
> 
> @mick711 Aidex support for Xdrip is also in the works, but was not yet merged. But until GlucoRx app supports it you won't be able to test it.

I am very much looking forward to its arrival and hope for early success