nightscout/AndroidAPS #1663: No notifikation for update from 2.8.2.1 to 3.0

NikieLiv, 2022-04-22:
First of all I want to apologize for the bumpy way of reporting, I have no experience with Github and how to report bugs.
And English is not my mother tongue.
I didn't get a message (pink) that it was now necessary to update from 2.8.2.1 to 3.0. I am using Android 9.
At 10:00 a.m. on April 21, the looper mobile phone suddenly showed only LGS and had stopped the closed loop.
Sorry, no log files because they can no longer be found on the mobile phone. (for me at least)
I just wanted to report that no update requests came this time.
Nik

Der-Schubi, 2022-04-22:
This is intended, if I'm not mistaken. 2.8.2.1 is only for use with older Android versions where 3.0 can't be used. Therefore the update notification has been disabled. In version 2.8.2 it's enabled.

dv1, 2022-05-11:
I propose then to add a different warning - if you use 2.8.2.1 on Android 9 or newer, show a warning that this version is NOT intended to be used with newer Android versions.

Der-Schubi, 2022-05-11:
Its not only the Android version, I think. Some people just want to wait with the update.

dv1, 2022-05-11:
@Der-Schubi  There's a 60 day grace period for the update. Plenty of time to wait. This isn't about that topic, it is about the grace period not being announced and silently running out. I agree with the developers that eventually forcing an update is absolutely necessary, and the 60 days are very generous. But this silent behavior with the 2.8.2.1 + Android 9+ combination is a problem.

NikieLiv, 2022-05-11:
@Der-Schubi Not in this case. I would have done the update if the system had told me to do it now. But without the prompt, I didn't know when the 60 days would start.

Der-Schubi, 2022-05-11:
Yes, not knowing when you have to update is part of the problem.