nightscout/AndroidAPS #1667: [dev-i] Bolus insulin not transferred to NSClient phone

kristinsmotz, 2022-04-22:
Bolused my daughter for a meal with the bolus calculator at 21.27 this evening and watched the bolus finish (including hearing the tics). In the AndroidAPS phone everything looks normal. In NSClient phone only the carb entry is visible, no bolus in graph or treatments tab. I believe I have seen this issue previously as well, but didn't grab logs that time. 
This time I managed to grab screenshots and logs from both phones. 
Nsclient: 
[image]
[image]
[AndroidAPS_LOG_1650658210138.log.zip](https://github.com/nightscout/AndroidAPS/files/8543903/AndroidAPS_LOG_1650658210138.log.zip)

AndroidAPS phone:
[AndroidAPS_LOG_1650658210138.log.zip](https://github.com/nightscout/AndroidAPS/files/8543911/AndroidAPS_LOG_1650658210138.log.zip)
[image]
[image]

MilosKozak, 2022-04-23:
try dev branch of NS

kristinsmotz, 2022-04-24:
I'll try that, but am running into issues since Heroku and Github are still not speaking to eachother. Had the same issue with breakfast bolus today, didn't show up in NSClient, only AAPS and NS. Did a full resync to the NSClient phone from NSClient tab, but didn't help. Will see if I can figure out another way to install the dev branch for nightscout.

kristinsmotz, 2022-04-26:
Haven't been able to update nightscout, but I did go to the phone app settings for NSClient on my phone on Sunday evening, cleared cache and data, and connected "clean" to NS again. It did not bring back the already missed bolus records, but I haven't had another missed bolus since. Now obviously can't trust the statistics in NSClient for a few days/up to a month, but that's relatively minor compared to not seeing boluses (and hassling daughter about not bolusing when she actually did bolus). Will update nightscout as soon as Heroku opens the GitHub connection, I apparently don't have it in me right now to figure out the direct Git text/code based interface update method.