nightscout/AndroidAPS #1672: [Wear] [GW4] [3.0.0] Bolus progress sound and/or vibration cannot be silenced

reijoahola-nightscout, 2022-04-24:
I have in Galaxy Watch 4 watch (GW4) notifications on and notification cause both sound and vibrate. However, I do not want these for AAPS bolus progress events (both sound and vibrate off for bolus progress).

There is AAPS setting in the watch "Vibrate on bolus" and I have turned it OFF but still it vibrates for bolus.

It would be also a nice feature to have a possibility to do the same for sound.

DM-LTR, 2022-04-24:
I am experiencing the very same behavior (Galaxy Watch 4, WearOS 3, AAPS master 3.0.0.1, app/wear apk fullrelease). Vibrate on bolus OFF, but every bolus step administered from smartphone is followed by vibration on watch (despite the watchface settings or watch sound/vibrate mode). This does not apply for individual SMBs. Turning "use system notifications" in AAPS settings does not make any change.

Andries-Smit, 2022-04-24:
This issue ia caused by the changed API for notification. Before wear last OS update is buzzed only twize. Before wear os verion x, the the notification could be changed. Fixing this issue removes the progress to the background. i dont want to do that, because the cancel button is not available. The work around is to build some thing custom.

vanelsberg, 2022-05-15:
@Andries-Smit How about a temporary work around by doing update notifications less often (e.g. every in progress steps of 25%)? This way on would only get 5 beeps, Cancel would still be possible.

Andries-Smit, 2022-06-02:
There is a workaround, for now, disabling the notification completely:
- Enable **Developer Settings** if you haven't already.
- About > Software > and keep tapping on **Software Version** until a toast appears.
- Back into Settings menu, **Developer options**
 - Scroll to bottom > App notifications > **Show all**
 - Select **AAPS**
 - Switch off the channel **AAPS Bolus Progress Silent**

 If you only switch off the single channel, the AAPS wear settings become effective.

In the Sttings of Watch AAPS
- Open **Menu**
- Select **Settings**
- Switch off **Vibrate on Bolus**

Choose AutoWear (and any other apps), select channels and voilà.

closed 2022-06-02 by MilosKozak

DM-LTR, 2022-06-03:
> * Scroll to bottom > App notifications > **Show all**

Tried... but this menu item is missing in my case (Android 12, Samsung One UI 4.1)

vanelsberg, 2022-06-06:
> ...this menu item is missing in my case (Android 12, Samsung One UI 4.1)

Note: this is about Android Wear OS (so on the watch, not the phone)

DM-LTR, 2022-06-06:
Ahh, sorry, but this was not clear at all. Watch settings is mentioned only after the 1st paragraph.

DM-LTR, 2022-06-26:
Solved in 3.1 beta 1...bolus delivery progress is announced via separate notification tile, not as ongoing/repeated one-time notification. Thx