nightscout/AndroidAPS #1680: Calibrations entered in XDrip cause NSClient to crash

Mikeparmenter11, 2022-04-26:
Hello, first time I've raised an issue.

When BG calibrations are entered in XDrip, it causes NSClient to crash.  This makes it very difficult to manage our 8 yr old T1 daughter remotely via NSClient, eg. when she is at school.

Sometimes the situation can be recovered by restarting dyno's in Heroku, but it can take several restarts and perhaps 10 minutes for NSClient to operate correctly.

This problem started after updating to AAPS v3.0.0.1

I have attached a screenshot of our NSClient logs, and Heroku logs.

Our rig set up is as follows:
- libre2, OOP2 to XDrip
- XDrip sends to AAPS (v3.0.0.1)
- AAPS uploads to Nightscout (v14.2.5)
- Xdrip followers (parents and grandparents)
- NSClient v3.0.0.1 used by parents and grandparents to correct carbs if school have entered them incorrectly, and do profile changes for PE etc.

It would be great if this issue could be resolved. Thank you for all your hard work, I don't know what we would have done without NS AAPS.

[NSclient crash after calibration error log.txt](https://github.com/nightscout/AndroidAPS/files/8566732/NSclient.crash.after.calibration.error.log.txt)

[image]

MilosKozak, 2022-04-29:
the sutiation is:
you calibrate in xdrip
calibration is passed to aaps(nsclient)
aaps upload calibration to NS
NS crash

correct?

MilosKozak, 2022-04-29:
could you provide logs from aaps from that time?

Mikeparmenter11, 2022-05-01:
Please find attached logs from straight after the crash.
KR 
Hannah

MilosKozak, 2022-05-02:
i see no logs

Der-Schubi, 2022-05-05:
I can confirm. My NS is down after I entered a bloody measurement.
I'll see if I can grab some logs.

MilosKozak, 2022-05-09:
no logs?

Der-Schubi, 2022-05-09:
Have not been able to reproduce it up to now.

MilosKozak, 2022-05-10:
fixed by https://github.com/nightscout/AndroidAPS/commit/6212bafcee0023fef452faa49b41eeb8ae9c6efa

closed 2022-05-10 by MilosKozak

Mikeparmenter11, 2022-05-10:
Apologies for my inexperience, I was reading your messages as emails, replying and attaching the logs to the emails, not realising they weren't getting to you / onto github.

Mikeparmenter11, 2022-05-10:
[AndroidAPS._2022-05-01_00-00-07_.1.zip](https://github.com/nightscout/AndroidAPS/files/8665031/AndroidAPS._2022-05-01_00-00-07_.1.zip)
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8665032/AndroidAPS.log)

MilosKozak, 2022-05-11:
hopefully i found the reason. you may test dev branch