nightscout/AndroidAPS #1682: Feature Request - Save TT-Checkbox state

goomStar, 2022-04-29:
I'd love to see AAPS saving the TT-checkbox state in the WizardDialog.

I personally am a fan of even making it "ON as default", because in my opinion every TT should be considered as intentional therapy decision and thus be automatically respected as default by the system when the user wants to bolus,
**BUT** I totally get the argumentation that humans are creatures of habit and every change makes them go "eeew" first.
This is why I'd appreciate to see at least the 'save state' feature, so that everyone benefits. Those who don't like change would'nt even notice and those who like their TT respected by default can live happily ever after <3

(I could elaborate why not respecting TT on bolus is a safety flaw in my opinion, but I leave that for now. Will go into detail if needed but I hope that a non-hurting change with benefits doen't need further details)

ga-zelle, 2022-09-01:
I like that proposal. One addition or independent feature is for active TT to prefill a new TT-Dialog with the current settings. This would be like an edit of the curent TT settings.

olorinmaia, 2026-10-06:
TT checkbox is now ON by default in v4 if TT is active. As I read this issue it can be closed.

closed 2026-10-06 by olorinmaia