nightscout/AndroidAPS #1683: Features : Split bolus over time

yrioux, 2022-05-01:
It would be great if we can split bolus over time. 

Usecase :
- In bolus wizard enter the amount of cards. 
- click on "split bolus".
- open new pop-up 
- select duration, split over x minutes 
- bolus from wizard is displayed. Click on it to split it in 2 parts. We could see percentage of each part. 
- can click on other parts to split again. 

This can be useful if you eat pizza and want to receive insulin in future. 

I also feel pain when bolusing more than 5 units, if I can split all my meals bolus over 30 minutes, it will be more confortable and convenient.

sebampuero, 2022-05-02:
If you would want to receive additional insuline as bolus or tbr in future for slow carbs or high fat content you would use eCarbs

yrioux, 2022-05-02:
But eCarbs only schedule carbs. I cannot schedule bolus...

I must to think to enter bolus later.

MilosKozak, 2022-05-02:
SMB is designed to do this automaticaly
Do a smaller bolus with food and let SMB do the rest 
Use ecarbs for for high fat foods

MilosKozak, 2022-05-02:
you can set % of given upfront bolus in preferences

nicoleloop, 2022-05-03:
I think that yrioux simply wanted the option to give an extended bolus that was unrelated to carb entries.

MilosKozak, 2022-05-09:
extended bolus goes directly against SMB alg. thus is not possible to use it

closed 2022-05-10 by MilosKozak