nightscout/AndroidAPS #1684: Feature request: allow automation to change ISF and IC without changing basal

JohnKitching, 2022-05-02:
I would like to change ISF and IC based on my BG. 
This is aligned with the (widely accepted) theory that ISF changes with BG. If ISF changes with BG then IC must too. 

There are ways of implementing similar functionality now but not without introducing other issues (that I can find)

Profile percentage change - the restrictions 70% to 130% are too tight and this method changes basal too 

Profile change - allows just changing ISF and IC but stops autosens working

I'd be happy with any functionality that allows a percentage change without changing basal or profile changes that don't reset autosens (not that making me happy is the goal here!) or a new way of making those changes with or without automation. 

I'd suggest it being an option rather than change in current functionality as I know not everyone accepts the theory.

Der-Schubi, 2022-05-03:
There is "Dynamic ISF" as plugin in the recent dev version. May you have a look at this?

JohnKitching, 2022-05-03:
> There is "Dynamic ISF" as plugin in the recent dev version. May you have a look at this?

I've been using that but there are a few problems with it (in my opinion):

It seems to undermine / works around some core AAPS functionality (autosens and insulin activity). 

It doesn't deal with Dynamic IC / CR. 

It doesn't seem to be going in a direction that results in it getting into NS/AAPS/master. 

All that said, the fact that dynamic ISF works extremely well for myself and many others is why I would like to see it built into core AAPS because it would be a game changer for so many people.

nicoleloop, 2022-05-03:
In my experience there may need to be some more thought about how to better incorporate carbs into AAPS, in general. This is a good a start as any to tackle this larger issue!

MilosKozak, 2022-05-09:
what we need is to simplify AAPS to allow use it by more people. things like this would have benefit for less than 1% of users
and it would make AAPS way more complicated in all areas: code, ui, understandability

closed 2022-05-09 by MilosKozak

JohnKitching, 2022-05-09:
Disagree completely but it's your football.