nightscout/AndroidAPS #1689: Urgency: Crash AAPS after update samsung One UI 4.1

vincent1890, 2022-05-03:
Good evening,
Following an update from samsung I can no longer use aaps I really need help urgently please

Reporting bugs
--------------
- 03/05/2022 a 20h01 Crash AAPS after update samsung One UI 4.1 (Android 12)
- 3.0.0.1 and version 3.0.0.2
- logs in attachments
[AndroidAPS._2022-05-03_01-20-12_.0.zip](https://github.com/nightscout/AndroidAPS/files/8613077/AndroidAPS._2022-05-03_01-20-12_.0.zip)
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/8613080/AndroidAPS.log)

vincent1890, 2022-05-03:
Excuse me all it’s been two hours that I was on it in the end after removing the cache and stopping force of the app I had the time a few seconds press the next button which allowed to unlock the situation and not crash the application anymore
Thank help @Robby in discord and WhatsApp

https://user-images.githubusercontent.com/5764518/166552998-808dde5a-0bac-4488-9d97-e22768ce59f9.mp4

[image]

vincent1890, 2022-05-03:
The video mounts the crash that was happening I don’t know if in the app there is a way to launch the app opened longer or have a bigger alert message about the permissions?

mick711, 2022-05-03:
Try to delete all files and folder on your phone, backup some files liké parameters. Delete database and uninstall aaps. Reboot thé phone install AAPS and set your parameters not with a backup.

MilosKozak, 2022-05-10:
resolved?

MilosKozak, 2022-05-18:
closed by inactivity

closed 2022-05-18 by MilosKozak