nightscout/AndroidAPS #1693: Since Update to Dev-i yesterday evening. Pump Connection Errors

Alamo04, 2022-05-04:
System:
- Dana RS pump
- Pixel 5 phone
- AAPS Dev-i (first with the update yesterday)

Since my update to dev-i (Build: 93af3d748) yesterday evening, I had 3 times "No connection to pump".
It seems that this message has nothing to do with the reality, because the pump was connected.

Logs attached.

[AndroidAPS_LOG_1651693367390.log.zip](https://github.com/nightscout/AndroidAPS/files/8626734/AndroidAPS_LOG_1651693367390.log.zip)

MilosKozak, 2022-05-05:
in logs everything looks normal, pump is connected withing a few seconds

Alamo04, 2022-05-05:
Last connection problem was yesterday 23:25.

So I think the problem was gone, perhaps with a new update and restart phone.

closed 2022-05-05 by Alamo04