nightscout/AndroidAPS #1694: Can't connect to dash pod anymore, need to change dash pod

golf4r, 2022-05-04:
Reporting bugs
Phone : Galaxy S22 Ultra
AAPS Version: 3.0.0.1
AAPS Build: 479bfc410-2022.02.07-17:33
Flavor : Full Release
Logs attached

Around 11:31 EDT, AAPS stopped working with Dash pods. This is the time where I jumped in the car and connected with Android Car for a Zoom meeting and GPS. I drove for around 30 minutes to get to my destination and AAPS never was able to connect back after. I had to come back to my house and sync a new pod. I tried restarting the phone, cleared cache and data in the phone, restarted bluetooth(from the system apps), nothing worked. 

The exact same issue happened last week when I used Android car again... had to sync a new pod when I came back home. I am attaching these logs as well for correlation, but it is possible they don't have the right time since I re-installed the application when I came back home since I did not know what to do.
[AndroidAPS._2022-04-30_13-36-16_.0.txt](https://github.com/nightscout/AndroidAPS/files/8627147/AndroidAPS._2022-04-30_13-36-16_.0.txt)

Please ping me for anything else that could be relevant and you need.
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
[AndroidAPS._2022-05-04_13-29-21_.txt](https://github.com/nightscout/AndroidAPS/files/8627110/AndroidAPS._2022-05-04_13-29-21_.txt)

- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-05-05:
this BT stack issue of your phone. AAPS cannot do anything with this issue.
Replace the phone or try factory reset

golf4r, 2022-05-05:
Hold on... can you point me to the evidence so I have something to tell Samsung for them to replace it ?

MilosKozak, 2022-05-05:
if it's a good phone try factory reset. It can be caused by connected BT device too

Mynez, 2022-05-30:
S9+ @golf4r haw did u resolve? I'll tried to delete all Bluetooth devices