nightscout/AndroidAPS #1697: PasswordReset not working

ofiryosifon, 2022-05-05:
April,2022

Resetting master password

PasswordReset file was put in AAPS/extra
And then restarting androidaps.
I entered Pod serial number in but without any help. Its not working..
Version 3.0.0
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-05-05:
pump?

MilosKozak, 2022-05-05:
for DASH password is 4241

ofiryosifon, 2022-05-06:
Thank :)
It work perfect

closed 2022-05-06 by MilosKozak