nightscout/AndroidAPS #1698: Wear ap stops connecting/synchronising after installing 3.0.0.2-dev-j  Commit#3d9b08e5

vanelsberg, 2022-05-05:
**AndroidAPS 3.0.0.2-dev-j**
**Build:** #3d9b08e5
**Flavor**: Fullrelease

_Wear App_: 2022.02.14 (Release version)
Build: #B1D45833C

**Problem description:**
With installing this build the Android Wear app stopped synchronizing.
APPS was updated at about 18:30h

App seems to not be connected,
Wear->RESEND DATA and Wear->OPEN SETTINGS ON WEAR have no effect.

[image]

[image]

** Log files **
[AndroidAPS_LOG_1651779641963.log.zip](https://github.com/nightscout/AndroidAPS/files/8635517/AndroidAPS_LOG_1651779641963.log.zip)

reijoahola-nightscout, 2022-05-05:
You have to update also Wear.

vanelsberg, 2022-05-05:
Expecting dev being compatible with release version. 
Apparently, it is not?

reijoahola-nightscout, 2022-05-05:
Not compatible in this case. There is some big reorganosation inside Wear dev-j.

reijoahola-nightscout, 2022-05-05:
I have both AAPS and Wear updated into dev-j but there seems to be mysterious problems what I do not understand. Could you update your systen and comment is it working without problems.

MilosKozak, 2022-05-05:
i see crashes comming from OnePlus Nord2 5G
caused by too old Wear OS

vanelsberg, 2022-05-05:
Current dev version

vanelsberg, 2022-05-05:
Upgraded Wear App to current dev. Yes, it is now showing syncing/BG.

Unfortunately for this version:
* Tiles are no longer functioning
* Watch faces layouts have "problems" (especially my favorite: "DigitalStyle")
* Watch settings (doubble screen-tap, settings) crashes when scrolling down)

So seems to be a work in progress...

vanelsberg, 2022-05-05:
> caused by too old Wear OS

Well, In my case an almost brand new Samsung Galaxy Watch4. Was working perfectly with tiles and all control functions.
Came sort of as a surprise as I expected release version of the Wear app to stay compatible.

MilosKozak, 2022-05-05:
tiles fixed, the rest is working well for me

MilosKozak, 2022-05-05:
[image]

vanelsberg, 2022-05-05:
> tiles fixed, the rest is working well for me

Ah, that's great!
Will test tomorrow and let you know my findings

vanelsberg, 2022-05-05:
On UI: using same watchface. 

- I can display date, including day name (as in THU, 5 MAY, 23:52)
It shows "52" (should be day name) and "52 MAY" (date)". Day name and date show "minutes?"

- Always on is also affected.

reijoahola-nightscout, 2022-05-06:
> On UI: I am using same watchface. Looks like UI is getting somewhat better now (layout may have been affected by database reset) But is still off on some parts.
> 
> * I can display date, including day name (as in THU, 5 MAY, 23:52) and week number
>   It shows "52" (should be day name) and "52 MAY" (date)". Day name and date show "minutes?"
> * Arrow on BG complication is not showing (something seems to be there)
> * BG complication shows "5." and then below something that looks like part of "3"
> * Not seeing basal & predeiction lines on graph
> 
> [image]
> 
> * Always on is also affected.
> 
> [image]
> 
> ** This is how release version looked **:
> 
> [image]

I have the same problems with the watchfaces. My watch is TicWatch Pro (from 2018) and the Wear OS should be the latest that the watch can update.

vanelsberg, 2022-05-06:
> > tiles fixed, the rest is working well for me
> Ah, that's great! Will test tomorrow and let you know my findings

Tried (current) dev build: On install AAPS started cycled crashes on startup.
So needed reverted back to previous build 3d9b08e5c2.

TT83TT, 2022-05-07:
for me, missing trend arrow and duplicate BG, OnePlus 8t phone and Galaxy Wear 4 watc
[image]
[image]
[image]
h

TT83TT, 2022-05-07:
New last commit 96f095ce8, still the same, missing arrow and double BG

MilosKozak, 2022-05-08:
https://github.com/nightscout/AndroidAPS/commit/e01647d496e405d3ae9617ae76f353147181ec02

MilosKozak, 2022-05-09:
https://github.com/nightscout/AndroidAPS/commit/b437307157420a8abcdf46bee5981ecae739df71
if there is still somthing else, open new issue

closed 2022-05-09 by MilosKozak

vanelsberg, 2022-05-09:
> [e01647d](https://github.com/nightscout/AndroidAPS/commit/e01647d496e405d3ae9617ae76f353147181ec02)

Yes, Looks ok now.
Thank for quick fixing: Much appreciated!