nightscout/AndroidAPS #1706: Negative global IOB on graph

robertrub, 2022-05-07:
[image]

[image]

Strangely on the graph of global IOB, the line is below zero...

AndroidAPS 3.0.0.2-dev-j 
Build: 16d39421b7-2022.05.07-14:30 

[image]

robertrub, 2022-05-07:
Some time later... The upper part of globsl IOB has been colored but the line is still below zero...

[image]

robertrub, 2022-05-07:
And a bit later...

[image]

ArthurusDent, 2022-05-07:
Duplicate of #1609 and #1597. 

Maybe sum up your but report into one post and add to #1597?

closed 2022-05-10 by MilosKozak