nightscout/AndroidAPS #1707: Strange watch face numbers and no arrow

robertrub, 2022-05-07:
[image]

AndroidAPS 3.0.0.2-dev-j 
Build: 16d39421b7-2022.05.07-14:30 

The watch app was built from dev-j. Can't remember if I built it from Tim's version or from Milos's dev.

rICTx-T1D, 2022-05-07:
I dont can reproduce it... please restart your wear watch and maybe your phone.

robertrub, 2022-05-08:
Tried, no change. All AAPS faces have the same problem, even the simple charging face.

We are both with French phones. Can there be an issue with the translation?

[image]

reijoahola-nightscout, 2022-05-08:
I have also the same "duplicate" problem, probably in all AAPS watcface types. The phone and watch have bees restarted several times. My watch is TicWatch Pro and Wear OS should be updated.

fjpezuela, 2022-05-08:
I have the same problem over here. I also use a TicWatch Pro 3 GPS 1853. The "Tiles" don't work either.

Last build: 3.0.0.2-dev-j 96f095ce8

robertrub, 2022-05-08:
@fjpezuela The tiles bug have been solved (?) in the last build. Did you build very recently ?

reijoahola-nightscout, 2022-05-08:
I have the latest updates except that 1d old Crowdin update. I stiil see such problems with Tiles that some of the tools does not open. Also that often mentioned BG duplicate problem is still a problem. My watch is TicWatch Pro (2018 year model) with the updated Wear OS. The language in the watch and phone is English.

robertrub, 2022-05-08:
@reijoahola-nightscout  You're correct about the tiles. They are visible but not functional :( Onlt the manual TT works on my TT tiles.

fjpezuela, 2022-05-08:
@robertrub Yes, only manual TT works. The other 3 don't work yet with the latest build on watch and phone.

vanelsberg, 2022-05-08:
Seeing the same.

**See https://github.com/nightscout/AndroidAPS/issues/1698:**

- Date display, including day name (as in THU, 5 MAY, 23:52) and week number:
It shows 52" (seconds, should be day name) and "52 MAY" (date)" (date show "minutes")

- Arrow on BG complication is not showing. Looks like it is showing the BG values wrapped to the next line?)

- Watch face settings (double-tap menue) crashes when scrolling to end

vanelsberg, 2022-05-08:
> @fjpezuela The tiles bug have been solved (?) in the last build. Did you build very recently ?

Only the "Actions" tile. Other are still not working.

MilosKozak, 2022-05-08:
https://github.com/nightscout/AndroidAPS/commit/e01647d496e405d3ae9617ae76f353147181ec02

fjpezuela, 2022-05-08:
The main screen already shows the trend date and the glucose value does not appear twice. TT Tiles are still not working.

reijoahola-nightscout, 2022-05-09:
Updated into the latest version. I have still this problem
- Watch face settings (double-tap menu) crashes when scrolling to end

MilosKozak, 2022-05-09:
> Updated into the latest version. I have still this problem
> 
> * Watch face settings (double-tap menu) crashes when scrolling to end

uninstall from watch and install again

reijoahola-nightscout, 2022-05-09:
> > Updated into the latest version. I have still this problem
> > 
> > * Watch face settings (double-tap menu) crashes when scrolling to end
> 
> uninstall from watch and install again

Thx, it corrected the setting problem.

robertrub, 2022-05-09:
It is working but I have no idea what the last part |-  should say... 

[image]

[image]

MilosKozak, 2022-05-09:
IOB fixed by https://github.com/nightscout/AndroidAPS/commit/7ef313e4aa22acd10571842f1dcfe36400c34fc7

robertrub, 2022-05-09:
@MilosKozak I confirm, you've solved it :) The watchface is working well now.

[image]

Next step, the tiles :)

MilosKozak, 2022-05-09:
fixed by https://github.com/nightscout/AndroidAPS/commit/b437307157420a8abcdf46bee5981ecae739df71

closed 2022-05-09 by MilosKozak