nightscout/AndroidAPS #1708: Filter prediction lines

sarahkovac, 2022-05-07:
I'd like to be able to choose which prediction lines show by default. Maybe the other prediction lines could be viewed in the graphs below or just selected in settings. I have been using this for months and still can't keep straight which line is which, and really there are only one or two that I need to see regularly. This would be much easier for beginners. Thanks!

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

yrioux, 2022-05-08:
For example, in xDrip, the prediction line is the combination of COB and IOB. It's also easier and faster to understand the graph with single prediction.

See line in purple 
[image]

sarahkovac, 2022-05-08:
Exactly. The Loop app on iPhone does a great job of this. I think AAPS is definitely smarter, but making the AAPS UI a little more user-friendly would go a long way. Having the predicted eventual glucose number would be super helpful as well.
[image]

> For example, in xDrip, the prediction line is the combination of COB and IOB. It's also easier and faster to understand the graph with single prediction.
> 
> See line in purple [image]

MilosKozak, 2022-05-09:
AAPS is different to loop
all prediction lines are used for decision thus keeping only one would not help to anything

closed 2022-05-09 by MilosKozak