nightscout/AndroidAPS #1711: Out of Whack Hardcoded Limits

NethCoder, 2022-05-09:
I have build 3.0.0.1. 

I noticed the DIA in the Profile Editor is Hard Coded to have a Minimum Value of 5.0. However, my Endocrinologist has specified mine should be 4.0. 

Also, in the Default Profile for the entire App, the Max Age is 19, and the Max Weight is 150.

Finally, and this is a *big* one, the Max Insulin I can Bolus is Hard Limited to 17 unit's. Frequently, that's OK; but, sometimes, I have to *try* to give myself more. The Calculator will say, for example 18.50 units (per current Glucose and Carbs). I can *try* and administered 1.50 units, but sometimes a will get an error saying I can't Bolus any more Insulin yet, and I'll have my choice of three buttons. Snooze Indefinitely, Snooze for 5 minutes, and OK. And, really the first and third choices are the same.

And, if I ever have a significant HIGH (meaning the CGM can't tell what my Glucose is, because it's over 400 mg/dL, I might have to use my backup: syringe & Insulin vial.

Can you fix these please?

I believe the DIA Range should be 2 to 5. In my 28 years of being a T1, I was at first told current Insulins, like novolog acted in 30min and were done in 2.5 hours. However, my current endocrinologist says 1 to 1.5 hours for peak effectiveness, and 4 hours until done; so, that's where that request is coming from.

I believe the age should go up to 110.
I believe the weight should go up to 300.

I believe the Insulin Bolus Max should be 30 units, because that's what my pump manufacture's Max is. So, that's an actual medical company opinion, not just mine. They're Insulet Co, makers of omnipod.com.

MilosKozak, 2022-05-09:
* DIA in loop perspective has different meaning than in classic pump therapy. 4h in DIA in pump has equivalent in about 8h in AAPS. Google internet for more info
* Default profile has data for children so far
* max bolus can be changed by selecting "higher" pacient type

NethCoder, 2022-05-10:
But if profile only has "children" data, how *do* I select a higher patient type. I already selected adult. The only higher is "pregnant." I'm male. Don't quire think those two things are compatible.

MilosKozak, 2022-05-11:
DIA:
https://seemycgm.com/2017/08/09/why-dia-matters/
https://www.diabettech.com/insulin/why-we-are-regularly-wrong-in-the-duration-of-insulin-action-dia-times-we-use-and-why-it-matters/

this is the only reason for age settings: 
https://androidaps.readthedocs.io/en/latest/Usage/Open-APS-features.html?highlight=pregnant#overview-of-hard-coded-limits

And you are mixing independent things together. Default profile is nothing more than a template in Profile helper. You may use it or not (and setup profile based on your experience). And like said before it has data only for children now.

KelvinKramp, 2022-05-17:
Hey neth, I understand your frustration, maybe you can adjust your dia in the code of your fork?

closed 2022-05-18 by MilosKozak