nightscout/AndroidAPS #1713: Wear app: Double-tab settings crash when scrolling down to end

vanelsberg, 2022-05-09:
**AndroidAPS 3.0.0.2-dev-j**
**Build:** #7ef313e4aa
**Flavor**: Fullrelease

Watch: Samsung Galaxy Watch4 44mm
**Wear App:** from AAPS 3.0.0.2-dev-j, build#7ef313e4aa
Build: #??????

**Problem description:** 

Watch face menu cashes on settings

_Repro_:

1. Double-tab on watch face for settings menu
2. Select "Settings"
3. Scroll down

When scrolling down beyond "Show Loop Status" the config page crashes back to the watch face.

MilosKozak, 2022-05-09:
uninstall wear app and install again

vanelsberg, 2022-05-09:
Ah... Missed that!
Tnx.

closed 2022-05-10 by MilosKozak

vanelsberg, 2022-05-12:
Had to uninstall/reinstall twice, but know it stopped crashing.

vanelsberg, 2022-05-12:
Menu started crashing again, now even worse disabling the entire watch face.
Reopening issue as https://github.com/nightscout/AndroidAPS/issues/1728