nightscout/AndroidAPS #1714: Wear app: BG arrow not visible on Watch Face AAPS(DigitalStyle)

vanelsberg, 2022-05-09:
**AndroidAPS 3.0.0.2-dev-j**
**Build:** #7ef313e4aa
**Flavor**: Fullrelease

Watch: Samsung Galaxy Watch4 44mm
**Wear App:** from AAPS 3.0.0.2-dev-j, build#7ef313e4aa
Build: #??????

**Problem description:** 

BG complication does not show direction arrow
Original WatchFace view (3.0.0.1 release):

[image]

MilosKozak, 2022-05-09:
what's there?
btw do you know how to add watchface supporting complications to Huawei Watch 2?

reijoahola-nightscout, 2022-05-10:
I have the latest version #58205e794. It shows BG arrow just like in that attached picture.
My watch is TicWatch Pro.

vanelsberg, 2022-05-10:
Will check latest build/version,

> btw do you know how to add watchface supporting complications to Huawei Watch 2?
Have aprox 0.1% experience with complications, do not have the Huawei Watch,
Would like to know, but - nope - sorry ;-)

vanelsberg, 2022-05-15:
@MilosKozak Fixed for mee.
Feel free to close..

closed 2022-05-18 by MilosKozak