nightscout/AndroidAPS #1721: Please increase the amount of days that log rotation keeps

ArthurusDent, 2022-05-11:
I'm in the middle of reporting some bugs and noticed that with 3.0.0.1 only about three days of logs are being kept anymore. That's very short when you don't immediately have time to report that bug. That's also quite a big change when taking up space on the smartphone wasn't a big issue before. I had hundreds of megabytes in logs. 

This also wasn't mentioned in the release notes, so some logs I needed for bug reporting are now gone.

I would suggest to keep the last two to four weeks.

MilosKozak, 2022-05-11:
there is 30 files kept. i believe it's enough

MilosKozak, 2022-05-18:
if there is rotation too fast , reduce logging amount. especially turn off autosense

closed 2022-05-18 by MilosKozak