nightscout/AndroidAPS #1724: Overlapping basals lead to incorrect IOB [Medtronic]

ArthurusDent, 2022-05-11:
Continuation of #1597 for Medtronic.

### I'll lead with the instance for which I actually have logs:

What can be seen here is a jump of IOB by ~ -0.4 units. I suspect that between 17:05 and 17:10 pump history had been read and records were updated. AndroidAPS did not know it had already set a zero temp earlier, which explains the jump in IOB. 
[image]

Pump history shows that AndroidAPS indeed set a zero temp twice.
[image]

Logs: 
[AndroidAPS._2022-05-09_16-33-16_.zip](https://github.com/nightscout/AndroidAPS/files/8670841/AndroidAPS._2022-05-09_16-33-16_.zip)

### The obligatory negative absolute insulin (no logs):

If I remember correctly, none of the 120 min zero temps actually ran the whole 120 min, I did stop them manually. Take a look at the two records I marked as A and B. The screenshots are only two minutes apart but the A-record was altered after the fact.

[image]

[image]

[image]

[image]

### System:
AndroidAPS 3.0.0.1 479bfc410
Oneplus 3T
Android 11 (LineageOS 18.1)
Medtronic 523
Dexcom G6 with BYODA

MilosKozak, 2022-05-11:
can you reset db and try dev?

ArthurusDent, 2022-05-12:
I'm on it.

ArthurusDent, 2022-05-13:
I've tried dev for about 18 hours and have made three observations.

#### 1. TBRs are not immediately stopped but they are stopped eventually

In the first screenshot you can see three TBRs that are running. In the second screenshot the two oldest TBRs have a stop time set. Also, IOB is different for these two TBRs when you compare the screenshots. In the third screenshot you can see that pump history was read and it seems that the records were updated and only then the TBRs were marked as stopped. I would guess this leads to slightly incorrect IOB in the meantime.
[image]
[image]
[image]

Logs:
[AndroidAPS-tbrs aren't stopped immediately.zip](https://github.com/nightscout/AndroidAPS/files/8686517/AndroidAPS-tbrs.aren.t.stopped.immediately.zip)

#### 2. Incorrect time left for TBR

For some reason time left for a TBR was incorrect in the pump tab.

[image]

[image]

[image]

[image]

[image]

[image]

Logs:
[AndroidAPS - incorrect time left for tbr.zip](https://github.com/nightscout/AndroidAPS/files/8686534/AndroidAPS.-.incorrect.time.left.for.tbr.zip)

#### 3. Looks like the algorithm didn't run for 10 min

I think AndroidAPS received all the bg values but the algorithm wasn't executed. 
[image]

[image]
[image]
[image]
[image]

Logs:
[AndroidAPS - algorithm didn't run.zip](https://github.com/nightscout/AndroidAPS/files/8686584/AndroidAPS.-.algorithm.didn.t.run.zip)

#### System:
AndroidAPS-dev ff46cbf19
Oneplus 3T
Android 11 (LineageOS 18.1)
Medtronic 523
Dexcom G6 with BYODA

ArthurusDent, 2022-05-13:
Just went through the past day and found one overlapping basal. Once again it was a manual suspension.

[image]

Logs:
[AndroidAPS dev-k overlapping.log](https://github.com/nightscout/AndroidAPS/files/8691728/AndroidAPS.dev-k.overlapping.log)

#### System:
AndroidAPS-dev https://github.com/nightscout/AndroidAPS/commit/ff46cbf1955de39daa3c0a48ead2521c4aee1286
Oneplus 3T
Android 11 (LineageOS 18.1)
Medtronic 523
Dexcom G6 with BYODA

MilosKozak, 2022-05-16:
ad 3) improved logging on different place
update and reset db pls
provide new logs

ArthurusDent, 2022-05-18:
Yesterday at around 6 pm I switched to 3d7298206.

A bad suspend happened that wasn't marked as stopped and the basals that were shown in the main screen were incorrect and pump history could not be synced. At least that's what it looked like to me.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

#### This is also still happening:

[image]
[image]

Logs:

[AndroidAPS pump history not synced.zip](https://github.com/nightscout/AndroidAPS/files/8715780/AndroidAPS.pump.history.not.synced.zip)

andyrozman, 2022-05-22:
i will take a look at this as soon as I can... This was already fixed, but it seems not fully... Somehow it works in the begining and then gets broken... (as I can see from snapshots)... 
I am pretty busy at the moment, I wanted to do it over this weekend, but I am affraid I couldn't...

andyrozman, 2022-05-29:
So I found the problem. 00:56, 30 minutes TBR is there, because we have't taken in account Rewind that happened few minutes afterwards. 
I hope to have it fixed soon.

andyrozman, 2022-05-29:
You can test the fix in this branch https://github.com/andyrozman/AndroidAPS/tree/bug1724_tbr_iob and we will get it merged as soon as I get your confirmation.

MilosKozak, 2022-05-31:
resolved in https://github.com/nightscout/AndroidAPS/pull/1790

ArthurusDent, 2022-06-05:
There is still something wrong. I've switched to 2d8e67e59 which should include Andy's patch. 

I suspended the pump at that time to do an infusion set change.

You can see overlapping basals:
[image]

[AndroidAPS log after fix.zip](https://github.com/nightscout/AndroidAPS/files/8839925/AndroidAPS.log.after.fix.zip)

andyrozman, 2022-06-05:
I will take a look as soon as I can, but probably not before next weekend... sorry. 

Maybe there is some mistake in my logic.

ArthurusDent, 2022-06-07:
I'm still on 2d8e67e and just witnessed that IOB calculation seems to be even more broken than before.

MaxIOB ist set to 1.5 and AndroidAPS thought there was more IOB and a few minutes later thought it was significantly less IOB. Reverting back to 3.0.0.2 master.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[AndroidAPS._2022-06-07_00-00-02_.6.zip](https://github.com/nightscout/AndroidAPS/files/8855697/AndroidAPS._2022-06-07_00-00-02_.6.zip)

andyrozman, 2022-06-07:
My branch shouldn't have that much impact (it just stopped running TBRs if
pump was Rewinding). But I will take a look at both situations... But this
is really weird, it seems like before we had first TBR fix applied (few
months ago, before 3.x was released).

ArthurusDent, 2022-06-07:
The rewind bug is an additional bug.

The original bug in this issue doesn't seem to have been fixed but correct me if you think it should have been fixed. I'm getting a bit lost with all these different reports.

andyrozman, 2022-06-07:
I actually thought that Rewind bug was the only bug. I didn't know there was different problem.

ArthurusDent, 2022-06-07:
The original bug seems to be overlapping basals (no rewind needed).

The rewind bug looks like a worse version of the original bug.

andyrozman, 2022-07-20:
Fix for such TBRs was done back in 3.0 before we had a first releases (same problem reported as you have), it seemed to be fixed then. In this fix I only extended the code a little so that we terminate TBR if Rewind was done afterward, but like ypu mentioned this doesn't fix all problems you have. 
I will need to do little bit more of deep-dive, to determine why this happens to you @ArthurusDent and in what situation this happens. Looking at the code this should work corectly, although I do need to extend the code to also include Pump Suspeds (to be able to terminate TBR).
I am currently not at home, but when I am back (next week) we can take another look and find the solution... Are you able to reproduce this at will?

ArthurusDent, 2022-07-21:
Preliminary remark: I have been using 3.0.0.2 (219bdba21) for six weeks now and it never showed IOB above my maxIOB. 

Right now I think that there are two possible reasons why IOB was wrong as described in this post: https://github.com/nightscout/AndroidAPS/issues/1724#issuecomment-1149030511

1. Dev https://github.com/nightscout/AndroidAPS/commit/2d8e67e5911469655b97cafdb8299cea8d20b5c1 has already had a bug that just became apparent in this situation.
2. Your fix caused the wrong IOB.

Could you try to explain to me why you think the problem has anything to do with a rewind (https://github.com/nightscout/AndroidAPS/issues/1724#issuecomment-1140440165)?

The problem is caused by a suspend that always precedes a rewind. Is the unsuspend after a rewind different from one without a rewind? 

What happens is that overlapping basals can be caused by any manual suspend and unsuspend, e.g. just to take a shower.

If I remember correctly the problem can be reproduced more or less reliably (without a rewind). No idea whether it can be reproduced more reliably with a rewind.

P.S.: It seems like the culprit with Omnipod was too low entropy. I've checked /proc/sys/kernel/random/entropy_avail. Entropy has been around 1000 in the last 12 hours.

andyrozman, 2022-07-24:
You are right current fix, actually makes things worse... I reverted the change, I will take a look at it later when I have time... 
I think I know what I need to do... I will take another look at it over the weekend...

andyrozman, 2022-08-06:
@ArthurusDent Could you please reproduce this with SUSPEND scenario and provide the logs... I started working on solution, but could not find sample in whole ticket include scenario when pump is suspended...

ArthurusDent, 2022-08-06:
Judging by the screenshots, the pump was suspended from 00:56 on here (reservoir change shortly after, as documented in one of the screenshots): https://github.com/nightscout/AndroidAPS/issues/1724#issuecomment-1129891406

Pump was uspended from 1:30 on here: https://github.com/nightscout/AndroidAPS/issues/1724#issuecomment-1146794240

Each one of those posts has logs included that show the specific time frame. Is something missing?

andyrozman, 2022-08-06:
Reservoir change is NOT the same as Suspend... 

When you do reservoir change, I need to look for Rewing record (there is no Suspend Pump triggered in that case) and when you do a real suspend, then I need to look for Suspend History entry...

ArthurusDent, 2022-08-06:
Oops, my bad. Incorrect terminology on my part. Replace every "suspend" with "disconnect".
[image]

I always do a disconnect before changing my reservoir or infusion set or to just let IOB fall as fast as possible.

I basically never use suspend.

andyrozman, 2022-08-20:
OK. I think this is now fixed... It seems that part of the code I added in 3.0, produced this error (I commented out part of code). With adding of handling of Suspend and Rewind records, this should now work correctly.

Can you please test it, and if it doesn't work send me logs and I will run it through the tests, to see if there is additional problem...

ArthurusDent, 2022-08-24:
I'm on it but it might take a bit of time.

andyrozman, 2022-08-24:
No problem with that from my side... As soon as it is tested, we can merge it back to dev, so that people can use it.

yp1ripe, 2023-05-03:
I am observing the same behavior as the original poster ( namely overlapping TBRs   lead to incorrect IOB ) with my daughter's 3.1.0.3. As I can see there's no  progress with the bug for about 8 months, so perhaps there are some work-arounds? Which releases are known to be free  from this bug? Any corrective actions mb? I've tried to delete some overlapping spans manually from my NS client and enter corrective TBRs from NS careportal with no significant luck ( my daughter's phone ignores NS-entered TBRs, at the same time mine accepts them). To some extent it helps to delete some TBR records on her phone while trying to achieve the smallest possible  difference with the externally (i.e. by means of autotune ) computed IOB. So, any comments?

ArthurusDent, 2023-05-03:
I wasn't comfortable with testing on a live system in the end. 

I'm currently in the process of getting a new pump so I hope that I will be able to start testing in four to six weeks.

You could go back to an older Android version that still runs AAPS 2.8.2.1, which is of course not ideal: https://androidaps.readthedocs.io/en/latest/Installing-AndroidAPS/Releasenotes.html#android-version-and-aaps-version

yp1ripe, 2023-05-04:
@ArthurusDent thank you for clarifications.

yp1ripe, 2023-05-10:
@andyrozman 

I believe that I found one bug in MedtronicHistoryData.kt

The problem is with 

    fun createTBRProcessList(entryList: MutableList<PumpHistoryEntry>) : MutableList<TempBasalProcessDTO>

Namely, the second "for" loop, ( the one  marked as         // fix for Zero TBRs ) doesn't take into account 
that  some number of zero temps may have their corresponding cancels already paired in the first "for" loop in the same function.

[zero_temps.patch](https://github.com/nightscout/AndroidAPS/files/11444642/zero_temps.patch)

I've started testing the patch with 3.1.0.3 and after half a day  I'd say it is much better than before.

MilosKozak, 2026-10-13:
The remaining case from the last comment is fixed on `dev` in a2308b5741.

`findDbEntry()` returned null in two cases: two temp-id entries close together ("Too many entries (with too small diff)"), and two candidates that were both more than 59 s away (its minute loop never widened the window). The TBR then got a new row by pumpId, and its temp-id row stayed open for the planned duration, so two basals overlapped. TBRs are now paired with the temp-id entries in one pass for the whole history read: the same rate first, as suggested above, then the smaller time difference.

Other fixes on `dev` from the same work, all about overlapping or wrong TBR rows on Medtronic:
- 181ec01caf a cancel ends the TBR that AAPS has running, not the last one read from history
- 84fe332e97 a new TBR ends the one it replaced (saving the new one did not end the old one)
- bbd9e440cd a TBR matched in history keeps its real length
- ba7d83c9f3 #5271, a re-read TBR no longer overwrites the next one
- b756aada44 and f95fb79390 a history read no longer fails at a page the pump does not have, or after a lost radio frame, so the repair from history really happens

Closing. Please reopen if overlapping basals still show on a build with these commits.

closed 2026-10-13 by MilosKozak