nightscout/AndroidAPS #1726: Wear OS DigitalStyle watchface time is missing minutes

tswire, 2022-05-12:
On the latest 3.0.0.2 dev k build 49163c43d, the digitalstyle watch face seems to have font issues (too big when set to AM/PM vs 24 hr) or missing info in the time.  See pic below.
[image]

vanelsberg, 2022-05-12:
Using 24 hrs format. For me is shows ok

tswire, 2022-05-12:
Must be the AM PM view only then. Hopefully someone can look at it as all these wear os faces were working perfectly before 3.0.0.2 build

jgslade, 2022-05-14:
I've also got the big font for the am/pm on digital style.
[image]

jgslade, 2022-05-14:
I was looking at the code to see how it all works (duct know Java/kotlin but have done some c, python, html, and sql).
I adjusted the font sizes in DigitalStyleWatchface.kt and adjusted the layout a bit on round/activity_digitalstyle.xml. It looks like the am/pm info is being passed with the hour info. 

[image]

jgslade, 2022-05-14:
I think I found a fix:

If this class
```
fun hourString(mills: Long): String {
        var format = "hha"
        if (android.text.format.DateFormat.is24HourFormat(context)) {
            format = "HH"
        }
        return DateTime(mills).toString(DateTimeFormat.forPattern(format))
    }
```

in
info/nightscout/androidaps/utils/DateUtil.kt

is changed to 
```
fun hourString(mills: Long): String {
        var format = "hh"
        if (android.text.format.DateFormat.is24HourFormat(context)) {
            format = "HH"
        }
        return DateTime(mills).toString(DateTimeFormat.forPattern(format))
    }
```

The display looks correct.
[image]

tswire, 2022-05-15:
> I think I found a fix:
>
> If this class
>
> ```
> fun hourString(mills: Long): String {
>         var format = "hha"
>         if (android.text.format.DateFormat.is24HourFormat(context)) {
>             format = [...]

Awesome work!!!  Thanks. 

How can this code have changed when it was working perfect before?  Are there controls in place to prevent something like this from happening?  Just curious.

jgslade, 2022-05-15:
Considering they just did a complete refactor of the wear code, it's not unheard of to have small bugs like this pop up.

tswire, 2022-05-15:
> Considering they just did a complete refactor of the wear code, it's not unheard of to have small bugs like this pop up.

What was the reason for this "refactor" of code?  Will it provide additional stability and benefits or functionality?  Curious

Der-Schubi, 2022-05-16:
> What was the reason for this "refactor" of code? Will it provide additional stability and benefits or functionality? Curious

Parts of the code were rewritten in Kotlin, before they were in Java.

MilosKozak, 2022-05-16:
main reson is the code was based on code from different sources including 8yo xdrip early versions, running on old google play libraries without possibility to upgrade without complete rewrite etc

closed 2022-05-16 by MilosKozak