nightscout/AndroidAPS #1727: Wear App: Tiles not communicating with app on phone

jgslade, 2022-05-12:
When I try to enter carbs using the AndoidAPS tiles on my Galaxy Watch 4 I no longer get the confirmation and the carbs aren't sent to the phone.

I did a demonstration at 12:36.

My phone is a Samsung s21FE the watch is a Galaxy Watch 4. I am using a dexcom g6 with BYODA and the Omnipod Dash. I am using an xdrip watchface. BYODA is sending data to both AAPS and xdrip, AAPS is uploading to nightscout and xdrip is uploading to tidepool.

Build number: ff46cbf19-2022.05.12-10:33 on the phone and FF46CBF19-2022.05.12 on the watch.

Video demonstarting the problem
https://user-images.githubusercontent.com/29897815/168148294-39e4ee4e-f21c-4f5c-a87c-1ca89495e4e8.mp4

Logs of the time surrounding the event.
[AndroidAPS_time surrounding event.log](https://github.com/nightscout/AndroidAPS/files/8681846/AndroidAPS_time.surrounding.event.log)

jgslade, 2022-05-12:
I saw on the discord that you need to reinstall the wear app, not just update. I have uninstalled and reinstalled the app. Since the tiles would work for me for a bit after installation on dev-j then stop I'll give it a few hours and if it's still working I close this ticket as an id10t error.

vanelsberg, 2022-05-12:
Known issue: full uninstall/reinstall is required.
For me tiles work OK for me so think your are fine now.

jgslade, 2022-05-12:
Just tried again. No longer getting the confirmation dialog and carbs aren't being sent to the phone.

I tried at 15:07 and again at 15:09
[AndroidAPS_time surrounding event 2.log](https://github.com/nightscout/AndroidAPS/files/8682622/AndroidAPS_time.surrounding.event.2.log)

Here's the full log as well as the cut down version.
[AndroidAPS (2).log](https://github.com/nightscout/AndroidAPS/files/8682627/AndroidAPS.2.log)

jgslade, 2022-05-12:
Restarted my watch and tried again and it worked.
I entered the carbs at 15:26

[AndroidAPS (3).log](https://github.com/nightscout/AndroidAPS/files/8682682/AndroidAPS.3.log)

I'll wait a couple of hours and see if it is still working.

jgslade, 2022-05-13:
Not working again.  I tried at 18:25. 

Tiles did work on dev-i without issue. On dev-j I experienced these same issues. 

[AndroidAPS (4).log](https://github.com/nightscout/AndroidAPS/files/8683242/AndroidAPS.4.log)

I guess the next step will be to rebuild the wear apk to see if that fixes it.

jgslade, 2022-05-13:
I've rebuilt the apk, and removed and replaced the wear app.

Same issues.

I also reset the watch. still having the same issues.

[AndroidAPS (5).log](https://github.com/nightscout/AndroidAPS/files/8689767/AndroidAPS.5.log)

FreDiabetics, 2022-05-13:
I have the exact same issue since AAPS V3.0 uninstalled and reinstalled wear apk before updating it. No confirmation and no treatments can be made.

jgslade, 2022-05-13:
Tried again and not getting the confirmation dialog anymore and the carbs aren't being sent to the watch. Tried the AAPS app instead of the tiles and it didn't work either this time, took the number but the confirmation dialog didn't pop up.

I started around 14:08
[AndroidAPS (6).log](https://github.com/nightscout/AndroidAPS/files/8691071/AndroidAPS.6.log)

From earlier it was about 12:45 ( don't remember the exact time I was at the park with my daughter.)
[AndroidAPS._2022-05-13_11-30-33_.1.zip](https://github.com/nightscout/AndroidAPS/files/8691081/AndroidAPS._2022-05-13_11-30-33_.1.zip)

jgslade, 2022-05-14:
I went to an AAPS watchface and the tiles appear to be working reliably now.  So it seems that the watchface is doing something with the connection with the phone that the tiles aren't doing on their own. 

Switched to the AAOS watchface around 11:40ish
[AndroidAPS._2022-05-14_00-00-06_.9.zip](https://github.com/nightscout/AndroidAPS/files/8693730/AndroidAPS._2022-05-14_00-00-06_.9.zip)

Foxy7, 2022-05-14:
I've found the same issue with Dev, that tiles not working correctly, but also that the calculator (previously called wizard) doesn't calculate insulin for a given amount of carbs, always returns zero uints.

Note, full uninstall of AAPS watch face and re-install didn't fix the issue, it still persisted.

Galaxy Watch 4.

vanelsberg, 2022-05-15:
On current build #FF46CBF195: Tiles are working just fine for me (Samsung Watch4 Classic)

vanelsberg, 2022-05-15:
> I've found the same issue with Dev, that tiles not working correctly, but also that the calculator (previously called wizard) doesn't  calculate insulin for a given amount of carbs, always returns zero uints.

@Foxy7 

In "double-tap settings", try setting "calculator percentage" to "ON"
Think this may be a bug. Waer app for 3.0 syncs the default percentage form AAPS settings. It does not seem to do that when setting is "OFF"?

vanelsberg, 2022-05-15:
@jgslade Install might be missing a privilege or power setting.
How are you installing the wear app to the watch?

jgslade, 2022-05-15:
I have tried installing it through adb from the computer and using the wear installer from my phone. I get the same results either way. I didn't change the settings on the phone or watch between dev i and dev k, except for when I reset the watch to see if that will help, then any settings that weren't backed up may not have been reset to where they were before.

vanelsberg, 2022-05-15:
> I have tried installing ...

Wear installer should be fine (using this myself). So that should not be the problem.

vanelsberg, 2022-05-15:
@jgslade 

> I am using an xdrip watchface....

Ah... you are running the xDrip watchface?
Tiles are part of the AAPS whatch face so the xDrip watch face wont work.
Try selecting the AAPS(DigitalStyle) watch face?

jgslade, 2022-05-15:
The tiles are working if I use an AAPS watchface, but when I use xdrip watchface they don't work, but they did work in dev i.

vanelsberg, 2022-05-17:
I think you cannot expect tiles designed for the AAPS watch face to work without it being active.

jgslade, 2022-05-17:
They did work with the xdrip watchfaces, but now they don't.  So if the goal is to get back the same functionality that existed before the translation to kotlin this is a legitimate bug.

MilosKozak, 2022-05-18:
current dev wear code is not compatible with xdrip watchfaces anymore (comm has changed)
You must use the same version of AAPS and WEAR to keep it working

closed 2022-05-18 by MilosKozak