nightscout/AndroidAPS #1728: Watch face settings menu (double-tab, settings) crashes Watch face when scrolling to the end.

vanelsberg, 2022-05-12:
**AndroidAPS 3.0.0.2-dev-j**
**Build:** #b4373071
**Flavor:** Fullrelease

**Watch:** Samsung Galaxy Watch4 44mm
**Wear App:** 2022.05.08

**Problem description:**

Watch face settings menu (double-tab, settings) crashes when scrolling to the end. After thsi sometime the watchface switches to "black" and can only be restored by restarting the watch.

**Additional note:**

After fully uninstalling and then reinstalling the wear app the problem solves, showing the settings menu.
But then after some time (event?) problem reappears.

See also https://github.com/nightscout/AndroidAPS/issues/1713

reijoahola-nightscout, 2022-05-13:
I believe that this is fixed in the latest dev version (3.0.0.2-dev-k 2022.05.12).

vanelsberg, 2022-05-15:
No crashing in the past 24 hrs. Yes, seems to be fixed on latest build: can not reproduce the problem.

@MilosKozak Feel free to close this issue. Will re-open when necessary...

jgslade, 2022-05-15:
I am experiencing this problem, I was trying to scroll down to try the workaround for the 0% issue on the calculator, but it crashes before I can get to the setting.

MilosKozak, 2022-05-18:
fixed by https://github.com/nightscout/AndroidAPS/commit/be199719bd92ac85a6d32587a839f3f728591f20
you need to uninstall/install if you are affected by this bug

closed 2022-05-18 by MilosKozak

vanelsberg, 2022-05-18:
Confirming, fixed!