nightscout/AndroidAPS #1729: Wear app: Calculator shows ISF value at the IC field.

vanelsberg, 2022-05-12:
**AndroidAPS 3.0.0.2-dev-j**
**Build:** #b4373071
**Flavor:** Fullrelease

**Watch:** Samsung Galaxy Watch4 44mm
**Wear App:** 2022.05.08

**Problem description:**

Calculator (from tile) shows ISF value at the IC field.

[image]

Foxy7, 2022-05-14:
Appears linked to #1727, i get the same with Dev that carbs entered get zero calculated insulin to be delivered.

jgslade, 2022-05-14:
Not sure if this fits best here or in its own ticket,  but since it's freaking with incorrect data from the calculator I thought it would be good here. 

It looks like for the dosage from the calculator it's multiplying it by 0% instead of the configured percentage. 

[image]

vanelsberg, 2022-05-17:
> ...not sure if this fits best here

Not related, but this already has a ticket:
https://github.com/nightscout/AndroidAPS/issues/1738

MilosKozak, 2022-05-18:
fixed by https://github.com/nightscout/AndroidAPS/commit/7254fc3654bb7e967b76b1d90fe25b72cd133fa3