nightscout/AndroidAPS #1732: Widget: minor arrow color bug

robertrub, 2022-05-14:
The sensitivity arrows are invisible.

[image]

osodebailar, 2022-05-14:
What you see is very dependent on how you set your background and the opacity of the widget .
With increasing the opacity you will see the arrow for example . With a lighter background you have the same effect

robertrub, 2022-05-14:
Before a recent update, those were visible with my transparent background.

robertrub, 2022-05-14:
Setting the background to opaque doesn't solve the problem either.

[image]

swissalpine, 2022-05-14:
I can confirm this issue. Opacitiy makes no difference.

osodebailar, 2022-05-14:
I will have a look on it in the next days

closed 2022-05-16 by MilosKozak