nightscout/AndroidAPS #1734: DASH: Updating to dev-k causes a lost POD

HuismanErik, 2022-05-14:
Today I updated AndroidAPS from dev-i to dev-k. After the update, there was no longer an active POD. Importing the backup settings didn't help either.

I reinstalled the old AndroidAPS APK and now the POD works again.
I would like to keep the active POD on an update.
Or should I update  AndroidAPS right before a POD switch?

robertrub, 2022-05-14:
This is not a bug. AAPS will wipe the pod history in order to clean the overlapping TBRs. You can either accept to lose the actual pod or you can use the pod with the "old" AAPS version, deactivate it. Install dev-k. Activate new pod.

You can close this issue if you don't have any other questions.

HuismanErik, 2022-05-15:
Ok thanks, I'll do the update right before a POD switch. I will close the issue.

closed 2022-05-15 by HuismanErik