nightscout/AndroidAPS #1735: accidental selection of insulin profile

timocm, 2022-05-15:
I discovered that my daughter's insulin profile was changed from NovaRapid to Lyumjev accidentally.

I guess it was when I was scrolling down in config builder that I touched a different insulin type.

Is it possible to have a screen in config builder, first showing the insulin profile and then if the insulin profile needs to be changed,  to go into an option that gives the various options of insulin types... or to have a confirmation pop-up to change insulin profile?

mick711, 2022-05-15:
You Can use parameters protection password, but it's a good Idea .

MilosKozak, 2022-05-15:
it's not possible because of internal program structure (all are plugins)

closed 2022-05-15 by MilosKozak