nightscout/AndroidAPS #1736: Objective 5 doesn't start

mushroom-dev, 2022-05-15:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
I completed objectives 1-4 and verified them, but objective 5 didn't event start.  This situation lasts since about 7 days.

- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
I found this issue in version 3.0.0.1. I tried to reinstall it few times, it didn't help. Today I upgraded to 3.0.0.2 (build 219bdba21-2022.05.15-10:19) and imported my settings, issue still persist.

- Obtain the app's log files, which can be found on the phone in
I've just installed 3.0.0.2 so my logs are empty ...

I've made all kind of stuff that was probably supposed to be done at this stage. I changed all the profile settings, used temp targets, went through all the `treatment safety` and `OpenAPS SMB` settings, especially I've set `Max U/h temp basal can set to` setting which I believe objective 5 is about.

MilosKozak, 2022-05-15:
can you provide screenshots?

mushroom-dev, 2022-05-16:
[image]

MilosKozak, 2022-05-16:
and screenshot of 1-3?

mushroom-dev, 2022-05-16:
[image]
[image]
[image]

Der-Schubi, 2022-05-16:
You have to verify them ^^

mushroom-dev, 2022-05-16:
Well .... it worked. I am pretty sure I clicked verify button before whatever it does. Anyway thank you

closed 2022-05-16 by mushroom-dev