nightscout/AndroidAPS #1741: [3.0.0.2-dev-k] "Error CGM data unchanged for the past ~45m..."

chrisbev, 2022-05-17:
[3.0.0.2-dev-k]  Got "Error CGM data unchanged for the past ~45m..." but data had changed over the 4.7 to 5.1mmol range, see attached.
NB using Tim2000 dISF latest.
Error cleared after about 30min.

[image]
[image]
[image]
[image]

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2022-05-17:
what CGM?

chrisbev, 2022-05-17:
FSL2 with patched LibreLink.  (NB Continuos yellow triangle).

chrisbev, 2022-05-17:
But it was reporting correct values AND they were changing? Please see the xDrip tab values for the 45mins?

chrisbev, 2022-05-17:
[AndroidAPS._2022-05-17_00-00-05_.1.zip](https://github.com/nightscout/AndroidAPS/files/8712185/AndroidAPS._2022-05-17_00-00-05_.1.zip)

chrisbev, 2022-05-17:
Log attached

chrisbev, 2022-05-18:
OK thanks for answer and time....

justmara, 2022-10-29:
OAPS check is wrong here. Because it checks long_delta. But the long delta would be zero for any period between two points of identical value. Thats how math works :) It you have two points of same value on graph - that means that sum of all changes between these points is zero. And this is exactly what average delta counts - it summarizes all the deltas and divides by count of elelements. It only shows the total change for period of time - difference between first and last point. But shows nothing about deviations. Bg could go +100..-100 for that period and total long_delta would be still `0`.
So this protection triggers when current BG is identical with 45min ago value, no matter what happened in between.
It must be checking for every BG point's variation and trigger only if all the points for last 45 (btw, why 45?) minutes and range of +-1 from current bg. This behaviour _could_ be considered suspicious.

ga-zelle, 2022-11-06:
Hi @justmara I agree that the check on long_avgdelta is not sufficient to detect changes in bg. However, it is one necessary requirement if bg was really flat.