nightscout/AndroidAPS #1742: Repeated vibration of Dana RS pump without obvious error

Foxy7, 2022-05-17:
Hi,
We're having repeated vibrations of one of our DanaRS pumps, the other one is working fine it seems.

The only thing i can see that is different is that we're seeing an announcement in the AAPS log of the vibrating pump called "Blood sugar measurement alert" at 07:38 yesterday day.

But the pump vibrated repeatedly this morning, so much so it woke me up, but nothing in the pump history via the LCD screen or in AAPS history either.

https://github.com/nightscout/AndroidAPS/blob/23207a275f97d1db1d993eae5122282920092602/danars/src/main/java/info/nightscout/androidaps/danars/comm/DanaRSPacketNotifyAlarm.kt#L46

If i read this correctly, it seems to be generated by the pump as an alarm. yet i don't think we've got BG reading set up on the pump.

Any ideas if there is a setting on the pump i need to turn off? maybe a battery issue?
thanks.

MilosKozak, 2022-05-17:
this is possible to disable in pump directly. AAPS only display that alarm

closed 2022-05-17 by MilosKozak

Foxy7, 2022-05-17:
Thanks, looks like it's a doctor mode factory setting i need to change on that pump.

Foxy7, 2022-05-18:
In case anyone else has this issue, it's related to Doctor mode and Melody option, the factory default for this pump was 120, but needs to be set to zero to turn off the alert which caused the vibration.

[image]