nightscout/AndroidAPS #1743: No syncing between NSClients -> AAPS

zwackelfuss, 2022-05-17:
I did not notice it first, but now I am sure:
Actions taken from NSClient are **not** synced to AAPS master, for example  entering carbs or setting a temp target like "eating soon" , although they are synced to Nightscout.

Syncing "the other way" from AAPS to NSClients and Nightscout works fine.

As using NSClient to handle the most common actions remotely without the need to have kid's AAPS phone in hands, I would be really happy if syncing would work properly again (as it did in 2.8, at least for us).

As always: thank you!

bchris21, 2022-05-17:
Same here, tested yesterday with Dev-K and nothing synced (carbs, TBR, etc).

MilosKozak, 2022-05-18:
i just tested it on latest dev and it works. i'd need an example with logs from both NSClient and master (latest dev version)

reijoahola-nightscout, 2022-05-18:
I tested this with the latest dev-k build 2022.05.16 and NSClient -> AAPS syncing worked.
Are you sure that you have enabled NSClient / Synchronisation settings in AAPS app Config Builder ?

MilosKozak, 2022-05-19:
I see there are 2 TTs with similiar time. there is a bug in NS. could you try it with dev version of NS?

bchris21, 2022-05-19:
Just tried with both Dev-k versions in NSClient and AAPS.

For me TBR is not passed to Nightscout at all.

**NSClient TBR 150mg/dl**
[image]

**NSClient not passing the TBR**
[image]

**Nightscout not reflecting the TBR**
[image]

**TBR not passed to AAPS**
[image]

[image]

[image]

MilosKozak, 2022-05-20:
I mean NS not NSClient

bchris21, 2022-05-26:
NS upgraded to Dev (14.2.6 - 555f699e) 

I canceled a TBR but still nothing is passed on NSClient. 

[image]

[image]

[image]

Any advice for further troubleshooting?

Thank you in advance. 

Chris

MilosKozak, 2022-05-26:
if you mean TT (not TBR) i tested now and it's working
guesing you don't have it enable in NSclient synchronization on AAPS side

bchris21, 2022-05-26:
Issue fixed for my case. 

Built and installed NSClient2 which works fine. Upgrading to NS Dev did not work.

Apparently something was corrupted on previous installation.

Again thanks for your support.

closed 2022-06-08 by MilosKozak