nightscout/AndroidAPS #1746: [Enhancement] SMS Commands from a non-smart phone

bchris21, 2022-05-18:
I work in an environment where smart phones are not allowed for security reasons. Phones without a camera, usually old models, are an exception to the rule.

I cannot use them though as I cannot take a photo of the QR-Code for 2FA. Is there a way not to use 2FA setup and use this type of phone? Maybe add support for a portable token generator.

Thanks

MilosKozak, 2022-05-18:
I don't know such device but for example Authy has application for PC

bchris21, 2022-05-18:
I know Yubico as hardware token that supports most well known services.

https://www.yubico.com/  

[image]

But I don't know how/if can this fit to AAPS and a non-smart phone.

I will meanwhile try to see how Authy web works.

Thanks Milos

MilosKozak, 2022-05-19:
yubico is for different purpose

closed 2022-05-19 by MilosKozak