nightscout/AndroidAPS #1749: [Safety] Stopping insulin delivery after bolus wizard logs carbs which then triggers delivery

szantos, 2022-05-19:
[v3.0.0.2 dev-k] but may affect master
May affect treatments button/quick wizard and any other situations where carbs and insulin are given/logged in one action.
See also #651

I've used the bolus wizard, entered carbs and pressed OK to bolus.
Then I've cancelled bolusing before injecting - no units have been delivered.
Carbs got logged.
Opening the wizard the carbs were there as COB - I have not reviewed this.
I've entered carbs again as new carbs and bolused.
Bolus was for COB + new carbs so basically double amount.

I understand, that I should have reviewed the details better before bolusing, and that this was mainly a user related error.

When the user cancels bolusing, it is to prevent insulin delivery. But if carbs get logged it triggers insulin delivery - this is the opposite. If a careless user input happens on top of it (like described above), bolusing for the double amount is possible.

Please reconsider this behaviour:
- carbs should not be logged 
- a message should show this 
- maybe if a partial injection has happened only that proportion of carbs could be logged (as a safe amount, but it makes this more complicated, it is easier without carb logging)

What are the benefits of logging carbs if stop is pressed at all?

MilosKozak, 2022-05-19:
in current version carbs are not logged if bolus is not successful (in oposition to previous versions)
if you experienced this in current dev, provide logs nad usual info

szantos, 2022-05-19:
Thank you for looking into it:
[https://drive.google.com/file/d/1JwcWkjalZr_RLjFymvnLfZhXWsS8QdIc/view?usp=drivesdk](url)

MilosKozak, 2022-05-19:
time?

szantos, 2022-05-20:
> time?

Yeah, sorry 12:20

MilosKozak, 2022-05-20:
12:20 is not in log file

szantos, 2022-05-20:
It was my last logfile, but it is from the day before it happened. I do not have a log of the day it happened at all. How can this be?