nightscout/AndroidAPS #1752: Feature Request: Change mechanism for determining SMB enablement

koelewij, 2022-05-20:
Now SMB is disabled for Libre 2 users, based on the assumption that the input data is too noisy to be used in combination with SMB's .
Within xDrip+ there are multiple noise suppression algorithms that generate a perfect smoot BG curve that is perfectly useful for SMB. 
On the other hand, i see curves of Dexcom G6 users that extend their sensors up to twice the lifetime of the sensor, that generates more noisy BG data than the Libre 2 and is allowed to keep on using SMB's with that data. 
Within OpenAPS SMB there is already an analysis of the noise inn the BG data available. Can this not be used to have SMB enabled instead of just looking at the type of sensor?

MilosKozak, 2022-05-20:
the issue is not only noisy sensor but situation where Libre sends false data (typicaly high value) instead of reporting error

koelewij, 2022-05-21:
I agree that we all should do our best to keep the loopers safe. But at a certain point it becomes the responsibility of the looper to decide if he wants to take the risk. Some serious warnings when a Libre user wants to activate SMB's are fine in my opinion also. A false high could just as easily mislead the bolus calculator and cause an overdose also.

MilosKozak, 2022-05-21:
current limitation affects typically only nights when people sleep and cannot recognise dangerous situation. i think it's good compromise

JohnKitching, 2022-05-23:
This request makes sense to me. 
I enable SMB always with Libre2 with a minor code edit so the risk is entirely mine. 
Would a "not recommended" or "beware of ..." label on the option suffice?

MilosKozak, 2022-05-23:
well if you get hurt or killed by this change nobody will ask if this was because your change. The reason for the world would be "dangerous AAPS". Think what will happen afterwards. 
So answer is no.
But it's not definive answer. When we find out (some/new) version of Libre is not affected and similiar reliable to Dexcom, we can change it.

closed 2022-05-23 by MilosKozak

harentius, 2022-12-18:
Just interested for myself: is Libre 3 more relieble? Or does it has the same issues as Libre 2?
With current integration approach, AAPS can't realise if data comes from Libre 2 or Libre 3 though.