nightscout/AndroidAPS #1754: Feature Request: Main Graph rendering for Profile Changes

koelewij, 2022-05-20:
The font and color used for showing a profile change in the graph is blurring the graph and also making other data hard to read. 
Changing the font and size to the same one that is used for Carbs & U of insulin, and giving it a different color (ie, magenta), would make it better to read and give less clutter on the screen. 
[image]

robertrub, 2022-05-21:
I "plus 1" this request. The graphics become unreadable and the information is shown twice (in orange and blue).
I think that @Philoul had already created a pull request to "clean up" the graph with a check box...

 N.B. Am I the only one having trouble to upload pictures here ??? After the 6th unsuccessful try, I give up :(

Philoul, 2022-05-21:
@robertrub My PR with checkbox has been replaced by an new item in graph menu to show or hide treatments.
Profile switches was not included in my old PR.

robertrub, 2022-05-21:
L[image]

Ok. I'd prefer to keep treatements and not see profil changes...

Philoul, 2022-05-21:
It's not recommanded to have so many profile switches (see https://androidaps.readthedocs.io/en/latest/Usage/Automation.html#incorrect-use-of-automation)
A looper should not have several PS per hours, but only PS for specific events (exercise, ...).

And active profile is the most important information of a looper (all calculated informations and commands done by loop algo use profile informations). That's why I didn't included PS in my PR.

We can of course improve presentation (font size and style, color), but I don't think hide this major information is a good idea...

robertrub, 2022-05-21:
@Philoul We can talk when your results will be better than mine. For the moment, I have 10% better TIR than you and only 1% hypos.

koelewij, 2022-05-21:
The profile % changes are based on the BG value. With Fiasp the efficiency gets less @ higher BG, so i compensate with higher profile %. I try to keep the temporary changes short, so it doesn't stay unnecessary long in a higher % than required. I might in the future step over on DynamicISF, but for the time being, this is my DynISF. Options to toggle treatment and profile changes in the graph would also help, but i personally like to see them but less dominant as they are now

MilosKozak, 2022-05-21:
improved with https://github.com/nightscout/AndroidAPS/commit/8b3c4c18c6a880cfb72fa8374e2be9aee03dcd4d

closed 2022-05-21 by MilosKozak